APS-USAXS/usaxs-bluesky-ended-2023 #505: WAXS detector priming

jilavsky, 2021-06-04:
I still got error, when we had to restart waxs ioc that detector HDF5 plugin is not primed. I thought this was solved by lazyopen as Mark suggested. Am I remembering this wrong?

prjemian, 2021-06-04:
That's the way Mark described it should work.

jilavsky, 2021-11-26:
Started instrument for December run, same error after restart of AD iocs:
```
subject: UnprimedPlugin during attempt 1 of 1 of command '('saxsExp', ['20', '21', '1', 'AgBehenate'], 23, '      saxsExp        20      21      1      "AgBehenate"')''

date: 2021-11-26 12:16:51.906081
command file: /share1/USAXS_data/2021-11/usaxs.mac
line number: 23
command: ('saxsExp', ['20', '21', '1', 'AgBehenate'], 23, '      saxsExp        20      21      1      "AgBehenate"')
raw command:       saxsExp        20      21      1      "AgBehenate"
attempt: 1 of 1
exception: The plugin hdf1 on the area detector with name saxs_det has not been primed.
Stopping further processing of this command list.
```
Solution is easy, simply collect image in each AD we have and all is done. But I wonder if this could be done by BS code when it finds unprimed state. SImply collect a data set to prime the thing and be done with it.