APS-USAXS/usaxs-bluesky-ended-2023 #507: BlackFly opticqal image is faling to collect data

jilavsky, 2021-06-15:
Nothing happens when trying. this is error message

```                                                                                                                                                                        
W Tue-10:52:40 - Could not take sample image:path=/share1/USAXS_data/2021-06/06_15_setupBS/06_15_setupBS_saxs, file=Blank, 
order#=9, exc=Attempted to set EpicsSignalWithRBV(read_pv='9idFLY2:JPEG1:Capture_RBV', name='blackfly_optical_jpeg1_capture', parent='blackfly_optical_jpeg1', value=0, timestamp=1623772350.0294104, 
auto_monitor=True, string=False, write_pv='9idFLY2:JPEG1:Capture', limits=False, put_complete=False) to value 1 and timed out after 10 seconds. Current value is 0.

D Tue-10:52:40 - areaDetectorAcquire(): OrderedDict([('enable', 1), ('auto_increment', 'Yes'), ('array_counter', 0), ('auto_save', 'Yes'), 
('num_capture', 0), ('file_template', '%s%s_%4.4d.hdf'), ('file_write_mode', 'Single'), ('blocking_callbacks', 'No'), 
('parent.cam.array_callbacks', 1)])
```
It is working manually from epics screen. I had to enable jpeg plugin and set to autosave, after which it saved image as expected.

prjemian, 2021-06-15:
@jilavsky : Was the `blackfly_optical` camera IOC just started?  This looks like the usual report from ophyd when trying to trigger an area detector IOC when the plugin has not already received at least one image with the current configuration.  (The so-called plugin *warmup*).

prjemian, 2021-06-15:
Can you report the value of `blackfly_optical.jpeg1.stage_sigs`?  (It's a dictionary.)

prjemian, 2021-06-15:
Or, the stage_sigs are specifying a configuration that the Capture value cannot accept.  One way to diagnose is (from the command-line), stage the detector and then inspect the configuration with the GUI screens.  Once satisfied that configuration correct, press `Acquire` to take an image.  Don't forget to unstage when done.

```python
blackfly_optical.stage()
# 1. check the GUI screens for correct values
# 2. press Acquire in the GUI
# 3. check for image on disk
blackfly_optical.unstage()
```

jilavsky, 2021-06-15:
I collected some data using spec and there it worked. Now it works in BS also. 
Looks like initialization from epics (without enabling jpg writer) is not good enough, spec enabled jpg writer, wrote image and disabled it. After this seems BS is working fine also. 
This initialization is nightmare. I have used the epics interface to find focus of the camera, it was collecting hundreds of images before I settled on focus and exposure. Yet, even after this, BS has initialization problem. This needs better solution on BS side, it cannot be this picky.  This is not usable for standard APS staff... 
I suggest to play with this on this camera after next Monday and finding solution for all AD BS code.

prjemian, 2021-06-15:
What you request is that BS do the same initialization steps as done by SPEC.  The problem is with area detector and how picky it is.

prjemian, 2021-06-15:
As [requested](https://github.com/APS-USAXS/ipython-usaxs/issues/507#issuecomment-861632371), 

> Can you report the value of `blackfly_optical.jpeg1.stage_sigs`? (It's a dictionary.)

jilavsky, 2021-09-14:
works now. Not sure why, but OK.

closed 2021-09-14 by jilavsky