APS-USAXS/usaxs-bluesky-ended-2023 #509: Check and verify command files

jilavsky, 2021-06-20:
We had user make typo in command file (usaxs.mac) and set sx=150 while correct should be sx=105. 150 was outside travel limits for sx and BS correctly identified this and stopped execution. BS sent e-mail, which was later received by staff and staff  remotely logged in, identified, fixed command file, and restarted. So this worked exactly as  intended - and considering this is running remote operations, I am happy... Yet another proof, that remote ops BS code works as intended... 
**Interesting** : this shows that even using Igor GUI based code for command generation & radiography, which is mostly button driven (but still allows typing) cannot solve this >20 years old issue of our users...  

**Request**: @prjemian help staff (JIL) to add verification of command files submitted. On submission, we should be able to verify the sx positions against sx limits and similarly we can check sy against its limits. If limits are breached, code should gracefully stop code execution and provide meaningful instructions in terminal. We know better, we can do better...

prjemian, 2021-06-21:
To evaluate the commands before they are run, you want to create the evaluation function and add it to: https://github.com/APS-USAXS/ipython-usaxs/blob/041d7d256c3f9952c6f1b6ad36f0f7a9267d75b6/profile_bluesky/startup/instrument/plans/command_list.py#L99-L117

Add the evaluation function just before the first `yield from bps.mv()` (before anything gets changed).  The `commands` object is a Python list of the commands.  The structure of a command in the list is: https://github.com/APS-USAXS/ipython-usaxs/blob/041d7d256c3f9952c6f1b6ad36f0f7a9267d75b6/profile_bluesky/startup/instrument/plans/command_list.py#L369-L380

Note: There's a typo in that excerpt above.  A `command` has these parts:

```rst
     command : tuple 
         (action, parameters, line_number, raw_command) 
```

The evaluation function should receive `commands` and loop through the list items, applying a check for each item **_only if_** it is appropriate (some items might not need to be checked).  Rather than _failing_ on the first problem discovered, the evaluation function collect any problems (error code found, raw command, line number) and proceed with checking the remaining items.

Once the list is checked, raise an exception of some sort (such as `RuntimeError`) and report any/all problems discovered.  If no problems, just return.  

Since this function is expected to complete its task quickly (and not delay execution of the bluesky RunEngine), it does not need to be written as a Bluesky plan.

prjemian, 2021-06-21:
Keep in mind, each command may have a different function with a different number of parameters and order of them.  Verify the call signature of each function type.  Here's a quick view:

```sh
usaxs@usaxscontrol ~/.../instrument/plans $ grep "def " scans.py
def preUSAXStune(md={}):
def preSWAXStune(md={}):
def USAXSscan(x, y, thickness_mm, title, md=None):
def USAXSscanStep(pos_X, pos_Y, thickness, scan_title, md=None):
def Flyscan(pos_X, pos_Y, thickness, scan_title, md=None):
def SAXS(pos_X, pos_Y, thickness, scan_title, md=None):
def WAXS(pos_X, pos_Y, thickness, scan_title, md=None):
```

This is the code that executes each command.  The evaluation code should have similar special handling for the different types of scans (for the different parameters):  https://github.com/APS-USAXS/ipython-usaxs/blob/041d7d256c3f9952c6f1b6ad36f0f7a9267d75b6/profile_bluesky/startup/instrument/plans/command_list.py#L435-L469

prjemian, 2021-06-21:
seems that something simple might handle all scans, such as:

```python
    scan_actions = "flyscan usaxsscan saxs saxsexp waxs waxsexp".split()
    if action.lower() in scan_actions:
        sx = float(args[0]) 
        sy = float(args[1]) 
        sth = float(args[2]) 
        snm = args[3] 
        # TODO: check sx against travel limits
        # TODO: check sy against travel limits
        # TODO: check sth for reasonable sample thickness value
        # TODO: check snm for reasonable sample title value
```

jilavsky, 2021-06-22:
@prjemian please check code in 509-verify-command-files branch. I tested it and it seems to fail correctly on violating low limit for sx and sy. Did not test high limits yet. 
Provide feedback and suggestions.

prjemian, 2021-06-22:
* naming style: change `ListOfErrors` to `list_of_errors`
* also record the line number and raw command (from the command file) so users can find the problem _easily_
* Python f-strings use lower case `f`, as in `f"An f-string {some_object}"`
* When you raise the error list, it should be formatted as a single string (hint: `"\n".join(list_of_errors)`).  Add a starting line that says errors were found in the command file.

prjemian, 2021-06-22:
https://namingconvention.org/python/variable-naming.html
[image]

jilavsky, 2021-06-22:
OK, fixed. 
output:
```
RuntimeError: Errors were found in command file. Cannot continue. List of errors:
SX low limit of -5.0 violated for sample Water Blank, USAXSscan, line number 19, command 		USAXSscan    -5       20     0      "Water Blank"
SY low limit of 20.0 violated for sample Water Blank, USAXSscan,  line number 19, command 		USAXSscan    -5       20     0      "Water Blank"
```

jilavsky, 2021-06-22:
@prjemian check again, ready to request merge? Anything else I am forgetting?

prjemian, 2021-06-22:
Much better.  Would it be informative to show the value of the limit?

Also, could remove the surrounding whitespace from the command, such as: `{raw_command.strip()}`

Perhaps reformat the error messages with line number first?

```
line 19: SX limit error: value (-5.0) below low limit (0), command: 'USAXSscan    -5       20     0      "Water Blank"'
```

PR #510 (2021-06-22): Verify command files
Fix #509
closed 2021-06-22 by jilavsky