APS-USAXS/usaxs-bluesky-ended-2023 #514: Need overwrite parameters method

jilavsky, 2021-09-13:
We want file BS_conf.py in current working directory in which we can overwrite selected parameters from their default values and potentially add new python functions for user use. Similar functionality to current USAXS_conf.mac and USAXS_user_macros.mac.
Preferable solution we planned: rite handling function which can handle each parameter which needs to be overwritten and empty overwrite structure in BS main code. 
Add file in working directory which will redefine this structure and call its handler function.

prjemian, 2021-09-13:
prototype back-end

```python
    elif 10.99 <= monochromator.dcm.energy.get() < 12.99:   # Si 220 crystals
        m_stage.r.tuner.width = 0.003
        a_stage.r.tuner.width = 0.0015
        m_stage.r2p.tuner.width = 9
        a_stage.r2p.tuner.width = 5
        ms_stage.rp.tuner.width = 3
        as_stage.rp.tuner.width = 3
        minstep = USER_OVERRIDE.get("usaxs_minstep", 0.000_035)
        yield from bps.mv(terms.USAXS.usaxs_minstep, minstep)
```

prjemian, 2021-09-13:
This could be initiated via a `%run -i file.py` or some other mechanism.  But the override structure must be available to the plans as they are written, so it needs some management.

prjemian, 2021-09-13:
A simple user-defined dictionary would not perform properly since the plan called from `instrument.plans.axis_tuning` would not see the same namespace.  Need the override structure to be managed from within the `instrument` package.

prjemian, 2021-09-13:
This begins to look like a Python class with attributes for each item to be supported with an override.

prjemian, 2021-09-13:
To be clear, user can change a value in an existing structure but user cannot re-create an existing structure.  (The new one will be in a namespace not recognized by the original.)

prjemian, 2021-09-13:
The code for this support should be usable by any device (which might have its own plan methods) or plan, without possibility of circular imports.  Of the existing subdirectories, `devices` is the next to be called after the Bluesky framework is initialized.  Put the override support in `instrument.devices`.

prjemian, 2021-09-13:
It will work like these test cases (with only support for `usaxs_minstep` parameter now).

### pick an unknown parameter

```ipython
In [3]: user_override.pick("elbows", "knees")
Out[3]: 'knees'
```

### pick a known parameter that has no override and a default value of `knees`

```ipython
In [4]: user_override.pick("usaxs_minstep", "knees")
Out[4]: 'knees'
```

### override the parameter and pick it again

```ipython
In [5]: user_override.usaxs_minstep = 2

In [6]: user_override.pick("usaxs_minstep", "knees")
Out[6]: 2
```

### reset the parameter and pick it again

```ipython
In [7]: user_override.usaxs_minstep = user_override.undefined

In [8]: user_override.pick("usaxs_minstep", "knees")
Out[8]: 'knees'
```

### What is the value of `undefined`?

```ipython
In [9]: user_override.undefined
Out[9]: <object at 0x7f387d443d80>
```

It appears to be random (it will be different in each python session).  Do not refer to it by value, just by name: `user_override.undefined`.

prjemian, 2021-09-13:
New `BS_conf.py` file:

```python
"""
Basic Bluesky configuration details for this installation of the USAXS instrument.

DO NOT USE TABS IN THIS FILE!

file: BS_conf.py
"""

print(__file__)

from instrument.devices import user_override

# override default values
user_override.usaxs_minstep = 0.000_01  # degrees, AR rotation minimum step
```

Load this from ipython command prompt:  `%run -i BS_conf.py`  (or `%run -im BS_conf`)

prjemian, 2021-09-13:
Here is the result, as used from a BS session:

```ipython
I Mon-17:45:14 - /home/beams11/USAXS/.ipython/profile_bluesky/startup/instrument/callbacks/nxwriter_usaxs.py
I Mon-17:45:14 - bluesky startup is complete

In [2]: user_override.summary()
============= =============
parameter     value
============= =============
usaxs_minstep --undefined--
============= =============

In [3]: user_override.usaxs_minstep = 0.000_01

In [4]: user_override.summary()
============= =====
parameter     value
============= =====
usaxs_minstep 1e-05
============= =====

In [5]:
```

prjemian, 2021-09-13:
Commit 93339b1 belongs to this issue.

prjemian, 2021-09-14:
Refactored the API so that `undefined` only used internally and does not appear in the user interface.  Refactor makes the process easier with a `register()` method, all known parameters are kept in an internal dictionary.

prjemian, 2021-09-14:
The `UserOverride` class would be an interesting addition to apstools.

prjemian, 2021-09-14:
Verified working as expected.

closed 2021-09-14 by prjemian