APS-USAXS/usaxs-bluesky-ended-2023 #515: BS cannot talk to BlackFly2 imaging camera?

jilavsky, 2021-09-21:
This is what we got:
```
W Tue-08:39:56 - Could not take sample image:path=/share1/USAXS_data/2021-09/09_21_Bavand/09_21_Bavand_usaxs, file=Sample_0, order#=743, exc=Attempted to set EpicsSignalWithRBV(read_pv='9idFLY2:JPEG1:FileTemplate_RBV', name='blackfly_optical_jpeg1_file_template', parent='blackfly_optical_jpeg1', value='%', timestamp=1632231586.2957368, auto_monitorTransient Scan ID: 24     Time: 2021-09-21 08:39:56plate', limits=False, put_complete=False) to value '%s%s_%4.4d.jpg' and timed out after 10 seconds. Current value is '%'.
Persistent Unique Scan ID: '7f49922d-3715-41df-bca6-647050203f36'
New stream: 'baseline'
New stream: 'aps_current_monitor'
```
BS was recently restarted and I can collect image manually in BlackFly2. When I collect manually, image appears in the right place.  BlackFly2 is OK, works from epics.

prjemian, 2021-09-21:
@jilavsky Can I operate this camera independent of your activities?

prjemian, 2021-09-21:
Just an idea:

> `path=/share1/USAXS_data/2021-09/09_21_Bavand/09_21_Bavand_usaxs, file=Sample_0`

You know that area detector needs a trailing directory separator (`/`) on the path?  Could that be the problem?

prjemian, 2021-09-21:
Why only 10 s for the timeout?

prjemian, 2021-09-21:
I'm seeing 44 such messages in the log file that spans  2021-09-18 11:50 to  2021-09-21 09:29

jilavsky, 2021-09-21:
You can operate this as you wish, impact on data collection is minimal. Spec may want to operate, but it writes PVs and is done. I can also disable spec collecting images
I wonder if I should restart ioc… but it worked, so I did not change anything. 
This worked until recently, so the question is what did we change?

jilavsky, 2021-09-21:
> Just an idea:
> 
> > `path=/share1/USAXS_data/2021-09/09_21_Bavand/09_21_Bavand_usaxs, file=Sample_0`
> 
> You know that area detector needs a trailing directory separator (`/`) on the path? Could that be the problem?

file path in epics does have trailing "/", it is possible this is problem.

prjemian, 2021-09-21:
Chances are it is not only writing this one PV at the time it fails.  Hardly reasonable for even a group of PVs to take as long as 1 second, let alone 10.

About the file path, when user writes a path to the write PV and does not include a trailing directory separator, area detector will add that automatically.  Bluesky fails in this case since it is waiting for the IOC to show that it has received the path _as requested_ (without the trailing separator).  

BUT, the fail reported is for `FileTemplate`.  I'm logged in now.

prjemian, 2021-09-21:
When you see the timeout in bluesky, the BlackFly detector is being staged.  This is the staging configuration:

```python
In [3]: blackfly_optical.jpeg1.stage_sigs
Out[3]:
OrderedDict([('enable', 1),
             ('auto_increment', 'Yes'),
             ('array_counter', 0),
             ('auto_save', 'Yes'),
             ('num_capture', 0),
             ('file_template', '%s%s_%4.4d.jpg'),
             ('file_write_mode', 'Single'),
             ('capture', 1),
             ('blocking_callbacks', 'Yes'),
             ('parent.cam.array_callbacks', 1)])
```

prjemian, 2021-09-21:
Ironically, the PV value is not changing so the timeout is even less important:

```python
In [4]: blackfly_optical.jpeg1.stage()

In [5]: blackfly_optical.jpeg1.unstage()

In [6]: blackfly_optical.jpeg1.file_template.get()
Out[6]: '%s%s_%4.4d.jpg'

In [7]: blackfly_optical.jpeg1.stage()

In [8]: blackfly_optical.jpeg1.file_template.get()
Out[8]: '%s%s_%4.4d.jpg'

In [9]: blackfly_optical.jpeg1.unstage()

In [10]: blackfly_optical.jpeg1.file_template.get()
Out[10]: '%s%s_%4.4d.jpg'
```

prjemian, 2021-09-21:
Taking an image with BS:

```python
In [1]: blackfly_optical.jpeg1.file_path.put("/mnt/share1/USAXS_data/2021-09/09_21_BS_test/")

In [2]: blackfly_optical.jpeg1.file_name.put("test")

In [3]: RE(bp.count([blackfly_optical]))

Transient Scan ID: 2     Time: 2021-09-21 16:37:39
Persistent Unique Scan ID: '65be23b7-f9da-4a4a-a0e3-91b7eceefc89'
New stream: 'baseline'
New stream: 'aps_current_monitor'
New stream: 'primary'
+-----------+------------+
|   seq_num |       time |
+-----------+------------+
|         1 | 16:37:41.7 |
+-----------+------------+
generator count ['65be23b7'] (scan num: 2)
Out[3]: ('65be23b7-f9da-4a4a-a0e3-91b7eceefc89',)

In [4]: !ls
20210921-163556.dat  test_0853.jpg

In [8]: !firefox test_0853.jpg &
```

[image]

prjemian, 2021-09-21:
test plan that uses the same code as the plan which USAXS runs:

```python
def bf_image():
    path = "/share1/USAXS_data/2021-09/09_21_BS_test"
    fname = "test"
    fnum = 1234
    yield from blackfly_optical.image_prep(path, fname, fnum)
    yield from blackfly_optical.take_image()
```

This worked as well: `RE(bf_image())`

prjemian, 2021-09-22:
Seems as if the additional sleep has solved this problem, not by waiting for unfinished `put()` operations to finish (there are none of those) but by allowing the `RE` to process other activities (respond to CA monitor events, allow for routine garbage collection by Python kernel, ...) before proceeding to `stage()` the `blackfly_optical` detector.  Otherwise, all these activities _appeared_ to cause a timeout for `blackfly_optical.jpeg1.file_template.put()`.

jilavsky, 2021-11-26:
November run. This issue is observed yet again:

```
W Fri-16:55:52 - Could not take sample image:path=/share1/USAXS_data/2021-11/11_26_Comissioning/11_26_Comissioning_waxs, file=AirBlank, order#=57, exc=Attempted to set EpicsSignalWithRBV(read_pv='9idFLY2:JPEG1:Capture_RBV', name='blackfly_optical_jpeg1_capture', parent='blackfly_optical_jpeg1', value=0, timestamp=1637967342.1474822, auto_monitor=True, string=False, write_pv='9idFLY2:JPEG1:Capture', limits=False, put_complete=False) to value 1 and timed out after 10 seconds. Current value is 0.
D Fri-16:55:52 - areaDetectorAcquire(): OrderedDict([('enable', 1), ('auto_increment', 'Yes'), ('array_counter', 0), ('auto_save', 'Yes'), ('num_capture', 0), ('file_template', '%s%s_%4.4d.hdf'), ('file_write_mode', 'Single'), ('blocking_callbacks', 'No'), ('parent.cam.array_callbacks', 1)])
```
Had to restart the ioc while BS was running since it was hanging even for epics operations. Restarted soft ioc and tested with epics, set exposure time. Checked we can operate device from epics and have displayed image in image viewer, but did not save. This resulted in "unable to collect image"

Next manually collected and **saved** jpg image --- and all started to work from BS without any other intervention. 

Looks like the camera needs to be run through save image procedure first???  Is this some priming issue for jpg image saver?  Can the issue of priming be resolved for all these AD cameras somewhere upstream on first failure?

closed 2022-02-21 by prjemian