APS-USAXS/usaxs-bluesky-ended-2023 #516: Change general parameters/PVs from run_command_file

jilavsky, 2021-09-22:
Sometimes we need to change scan parameters in between samples. Example: slit sizes, SAXS/WAXS exposure time, fly scan time, number of steps for step scan,... Example, last night I had 5 samples and 3 needed small beam while two could have used large beam. I had to run all with small beam... 
These are PVs in epics linked to BS parameters, typically in general_terms.py. 
In spec we have bunch of functions (e.g.: set_USAXS_MINSTEP) which one can call in run_command_file when we need to change that specific value. 
Looking at comand_list.py, I do not see this in actions available in run_command_file. It would be nice to implement suitable method how to modify these.
These are really not suitable for our overwrite method, since there is no defaults in BS. BS is using current epics value and has no default values for these parameters. Can we add some simple action able to do this? Any other suggestion?

prjemian, 2022-01-28:
Thinking that we could add a `Set` command to the command handler:  https://github.com/APS-USAXS/ipython-usaxs/blob/7affdc1c9cd7f4738112d7e6c068fbc30dad9768/profile_bluesky/startup/instrument/plans/command_list.py#L482

and maybe this line, too: https://github.com/APS-USAXS/ipython-usaxs/blob/7affdc1c9cd7f4738112d7e6c068fbc30dad9768/profile_bluesky/startup/instrument/plans/command_list.py#L156

The new `Set` command (in `usaxs.mac`) would look like these examples:

```text
Set terms.SAXS.usaxs_h_size 0.2
Set terms.SAXS.usaxs_v_size 0.2
```

We could/should limit this to _only_ items in `terms`.  Agree?  Would not be too difficult.  Values would be set via `yield from bps.mv(object, value)` where `object` must be constructed from the text of `terms.SAXS.usaxs_h_size`.

jilavsky, 2022-01-30:
I agree that Set command would be suitable. A:so. limit to terms is fine - at least for now. We will see (and implement if really needed) any other parameters, but terms should be all we need.

prjemian, 2022-01-31:
OK, ready to test.  Here are some examples calling the lower-level code directly.  The first few are correct, the remainder show some error handling:

```python
In [112]: RE(run_set_command("terms.SAXS.usaxs_v_size", "0.8")); terms.SAXS.usaxs_v_size.get()
Out[112]: 0.8

In [113]: RE(run_set_command("terms.SAXS.usaxs_v_size", "0.802")); terms.SAXS.usaxs_v_size.get()
Out[113]: 0.802

In [114]: RE(run_set_command("terms.SAXS.usaxs_v_size", .8)); terms.SAXS.usaxs_v_size.get()
Out[114]: 0.8

In [115]: RE(run_set_command("terms.SAXS.usaxs_v_size", "chuck")); terms.SAXS.usaxs_v_size.get()
E Mon-12:38:55 - Cannot set 'terms.SAXS.usaxs_v_size' to 'chuck' due to exception (could not convert string to float: 'chuck').  Skipping this command ...
Out[115]: 0.8

In [116]: RE(run_set_command("terms.SAXS.usaxs_v_size.get", "chuck")); terms.SAXS.usaxs_v_size.get()
E Mon-12:39:10 - Cannot set 'terms.SAXS.usaxs_v_size.get', it is not a Signal.  Skipping this command ...
Out[116]: 0.8

In [117]: RE(run_set_command("terms.SAXS.usaxs_vertical_size", "chuck")); terms.SAXS.usaxs_v_size.get()
E Mon-12:39:27 - 'terms.SAXS.usaxs_vertical_size' does not exist.  Skipping this command ...
Out[117]: 0.8

In [118]: RE(run_set_command("terms.SAXS", "chuck")); terms.SAXS.usaxs_v_size.get()
E Mon-12:39:36 - Cannot set 'terms.SAXS', it is not a Signal.  Skipping this command ...
Out[118]: 0.8

In [119]: RE(run_set_command("RE", "chuck")); terms.SAXS.usaxs_v_size.get()
E Mon-12:39:45 - First argument must start with 'terms.', received 'RE'.  Skipping this command ...
Out[119]: 0.8

In [120]: RE(run_set_command("terms.SAXS", "one", "two")); terms.SAXS.usaxs_v_size.get()
E Mon-12:41:29 - syntax:  SET terms.component value, received ('terms.SAXS', 'one', 'two').  Skipping this command ...
Out[120]: 0.8

In [121]: RE(run_set_command("terms.SAXS.will.not.be.found", "one", "two")); terms.SAXS.usaxs_v_size.get()
E Mon-12:42:39 - syntax:  SET terms.component value, received ('terms.SAXS.will.not.be.found', 'one', 'two').  Skipping this command ...
Out[121]: 0.8

In [122]: RE(run_set_command("terms.SAXS.will.not.be.found", "one")); terms.SAXS.usaxs_v_size.get()
E Mon-12:42:49 - 'terms.SAXS.will' does not exist.  Skipping this command ...
Out[122]: 0.8

```

prjemian, 2022-01-31:
Here is an example command file:

```text
# test the `set terms.thing value` command

# valid syntax, no such attribute
set terms.thing value

# too many arguments
set anything value1 value2

# does not set an attribute of `terms`
set anything value1

# bad value
SET terms.SAXS.usaxs_v_size chuck

# example from GH issue of valid commands
Set terms.SAXS.usaxs_h_size 0.2
SET terms.SAXS.usaxs_v_size 0.802
```

prjemian, 2022-01-31:
Can't actually test this now, it seems:

<details>

```python
In [124]: summarize_plan(run_command_file("dev_test_set.txt"))
I Mon-12:48:16 - Command file: dev_test_set.txt
====== ====== ==============================
line # action parameters                    
====== ====== ==============================
4      set    terms.thing, value            
7      set    anything, value1, value2      
10     set    anything, value1              
13     SET    terms.SAXS.usaxs_v_size, chuck
16     Set    terms.SAXS.usaxs_h_size, 0.2  
17     SET    terms.SAXS.usaxs_v_size, 0.802
====== ====== ==============================

I Mon-12:48:16 - memory report: pmem(rss=327823360, vms=4239720448, shared=49004544, text=2560000, lib=0, data=3720949760, dirty=0)
=================================== Open Run ===================================
  Read ['text']
================================== Close Run ===================================
I Mon-12:48:16 - Command file verified
user_data_time_stamp -> 2022-01-31 12:48:16.374160
user_data_state -> Starting data collection
user_data_collection_in_progress -> 1
usaxs_shutter -> close
terms_SAXS_collecting -> 0
terms_WAXS_collecting -> 0
a_shutter_autoopen -> 1
I Mon-12:48:16 - Measuring background for: PD_USAXS
scaler0_preset_time -> 0.2
scaler0_auto_count_delay -> 0
upd_autorange_controls_mode -> manual
I0_autorange_controls_mode -> manual
I00_autorange_controls_mode -> manual
trd_autorange_controls_mode -> manual
upd_autorange_controls_reqrange -> 4
I0_autorange_controls_reqrange -> 4
I00_autorange_controls_reqrange -> 4
trd_autorange_controls_reqrange -> 4
upd_autorange_controls_ranges_gain4_background -> 1.0
upd_autorange_controls_ranges_gain4_background_error -> 0.0
I Mon-12:48:16 - PD_USAXS range=4 gain=1e4 bkg=1.0 +/- 0.0
I0_autorange_controls_ranges_gain4_background -> 4.0
I0_autorange_controls_ranges_gain4_background_error -> 0.0
I Mon-12:48:16 - I0_USAXS range=4 gain=1e5 bkg=41.2 +/- 0.4
I00_autorange_controls_ranges_gain4_background -> 1.0
I00_autorange_controls_ranges_gain4_background_error -> 0.0
I Mon-12:48:16 - I00_USAXS range=4 gain=1e5 bkg=1.0 +/- 0.0
trd_autorange_controls_ranges_gain4_background -> 1.0
trd_autorange_controls_ranges_gain4_background_error -> 0.0
I Mon-12:48:16 - TR diode range=4 gain=1e5 bkg=3768.8 +/- 8.953211714239757
upd_autorange_controls_reqrange -> 3
I0_autorange_controls_reqrange -> 3
I00_autorange_controls_reqrange -> 3
trd_autorange_controls_reqrange -> 3
upd_autorange_controls_ranges_gain3_background -> 1.0
upd_autorange_controls_ranges_gain3_background_error -> 0.0
I Mon-12:48:16 - PD_USAXS range=3 gain=1e4 bkg=1.0 +/- 0.0
I0_autorange_controls_ranges_gain3_background -> 4.0
I0_autorange_controls_ranges_gain3_background_error -> 0.0
I Mon-12:48:16 - I0_USAXS range=3 gain=1e5 bkg=7.8 +/- 13.6
I00_autorange_controls_ranges_gain3_background -> 1.0
I00_autorange_controls_ranges_gain3_background_error -> 0.0
I Mon-12:48:16 - I00_USAXS range=3 gain=1e5 bkg=1.0 +/- 0.0
trd_autorange_controls_ranges_gain3_background -> 1.0
trd_autorange_controls_ranges_gain3_background_error -> 0.0
I Mon-12:48:16 - TR diode range=3 gain=1e5 bkg=345.4 +/- 1.3564659966250536
upd_autorange_controls_reqrange -> 2
I0_autorange_controls_reqrange -> 2
I00_autorange_controls_reqrange -> 2
trd_autorange_controls_reqrange -> 2
upd_autorange_controls_ranges_gain2_background -> 1.0
upd_autorange_controls_ranges_gain2_background_error -> 0.0
I Mon-12:48:16 - PD_USAXS range=2 gain=1e4 bkg=1.0 +/- 0.0
I0_autorange_controls_ranges_gain2_background -> 4.0
I0_autorange_controls_ranges_gain2_background_error -> 0.0
I Mon-12:48:16 - I0_USAXS range=2 gain=1e5 bkg=34.6 +/- 0.4898979485566356
I00_autorange_controls_ranges_gain2_background -> 1.0
I00_autorange_controls_ranges_gain2_background_error -> 0.0
I Mon-12:48:16 - I00_USAXS range=2 gain=1e5 bkg=1.0 +/- 0.0
trd_autorange_controls_ranges_gain2_background -> 1.0
trd_autorange_controls_ranges_gain2_background_error -> 0.0
I Mon-12:48:16 - TR diode range=2 gain=1e5 bkg=1.0 +/- 0.0
upd_autorange_controls_reqrange -> 1
I0_autorange_controls_reqrange -> 1
I00_autorange_controls_reqrange -> 1
trd_autorange_controls_reqrange -> 1
upd_autorange_controls_ranges_gain1_background -> 1.0
upd_autorange_controls_ranges_gain1_background_error -> 0.0
I Mon-12:48:16 - PD_USAXS range=1 gain=1e4 bkg=1.0 +/- 0.0
I0_autorange_controls_ranges_gain1_background -> 4.0
I0_autorange_controls_ranges_gain1_background_error -> 0.0
I Mon-12:48:16 - I0_USAXS range=1 gain=1e5 bkg=11.6 +/- 13.821722034536796
I00_autorange_controls_ranges_gain1_background -> 1.0
I00_autorange_controls_ranges_gain1_background_error -> 0.0
I Mon-12:48:16 - I00_USAXS range=1 gain=1e5 bkg=1.0 +/- 0.0
trd_autorange_controls_ranges_gain1_background -> 1.0
trd_autorange_controls_ranges_gain1_background_error -> 0.0
I Mon-12:48:16 - TR diode range=1 gain=1e5 bkg=1.0 +/- 0.0
upd_autorange_controls_reqrange -> 0
I0_autorange_controls_reqrange -> 0
I00_autorange_controls_reqrange -> 0
trd_autorange_controls_reqrange -> 0
upd_autorange_controls_ranges_gain0_background -> 1.0
upd_autorange_controls_ranges_gain0_background_error -> 0.0
I Mon-12:48:16 - PD_USAXS range=0 gain=1e4 bkg=1.0 +/- 0.0
I0_autorange_controls_ranges_gain0_background -> 4.0
I0_autorange_controls_ranges_gain0_background_error -> 0.0
I Mon-12:48:16 - I0_USAXS range=0 gain=1e5 bkg=35.0 +/- 0.0
I00_autorange_controls_ranges_gain0_background -> 1.0
I00_autorange_controls_ranges_gain0_background_error -> 0.0
I Mon-12:48:16 - I00_USAXS range=0 gain=1e5 bkg=1.0 +/- 0.0
trd_autorange_controls_ranges_gain0_background -> 1.0
trd_autorange_controls_ranges_gain0_background_error -> 0.0
I Mon-12:48:16 - TR diode range=0 gain=1e5 bkg=1.0 +/- 0.0
scaler0_preset_time -> 0.025
scaler0_auto_count_delay -> 0.02500000037252903
TypeError: '<' not supported between instances of 'KohzuSoftPositionerTuple' and 'float'

In [125]: 
```

</details>

Possibly try tomorrow, suspecting this is a shutdown consequence.

prjemian, 2022-01-31:
This is not ready.  It only allows `Signal` (or subclass) to bet set.  Does not work with positioners yet.

prjemian, 2022-01-31:
If it is a positioner, test for `hasattr(obj, "move")` and get object's value from `obj.position`.

jilavsky, 2022-01-31:
This should NOT be moving anything. These are values/PVs for slit sizes (USAXS, SAXS/WAXS) and we are changing these values/PVs. USAXS changes slit sizes all the time. We are rewriting the settings used for USAXS or SAXS/WAXS. We are NOT moving anything. 
None of the terms I need to change should move motors when changed. I want to change conditions used set by mode* commands when we go between geometries.

prjemian, 2022-01-31:
Ok, added generous comment describing our intentions and continued use of `bps.abs_set()` (no wait for completion).  Retained all the input error checking since it protects us from our own typos in the command files.

prjemian, 2022-01-31:
Here is example of a pure Python (non-EPICS) Signal:

```python
In [21]: RE(run_set_command("terms.FlyScan.setpoint_up", 1000));  time.sleep(.1); terms.FlyScan.setpoint_up.get()
Out[21]: 1000

In [22]: RE(run_set_command("terms.FlyScan.setpoint_up", 6000));  time.sleep(.1); terms.FlyScan.setpoint_up.get()
Out[22]: 6000
```

PR #543 (2022-01-31): add SET command for batch execution
- FIX #516
closed 2022-01-31 by prjemian