APS-USAXS/usaxs-bluesky-ended-2023 #517: Switch A shutter AutoOpen on

jilavsky, 2021-09-24:
We need to make sure 9ida:AShtr:Enable is set to 1 when we collect data. We need to add this command somewhere where it is conveniently run. I added this to spec to CollectData command. It should be added in BS when we start BS and are NOT in simulation mode (when C station is enabled). I will investigate and propose solution...

prjemian, 2021-09-24:
You could add it to https://github.com/APS-USAXS/ipython-usaxs/blob/dbdf3f2dd9c00e825b2e84496dff73ffd57d5661/profile_bluesky/startup/instrument/plans/command_list.py#L100 or https://github.com/APS-USAXS/ipython-usaxs/blob/dbdf3f2dd9c00e825b2e84496dff73ffd57d5661/profile_bluesky/startup/instrument/plans/command_list.py#L201

Don't forget to change it back to your desired setting in the associated (whichever `before_` plan you pick) `after_` plan.

jilavsky, 2021-09-24:
It stays as 1 (Enabled). We want this on be default and usually staff will do it when beam comes back after 1 day shut down. We sometimes forget and then loose beam (like tonight). There is no need to set back to 0, ever.

jilavsky, 2021-09-24:
Here is my thought on this. Add somewhere 
```
from ophyd import EpicsSignal
a_shutter_autoopen = EpicsSignal(
    "9ida:AShtr:Enable",
    name="a_shutter_autoopen")
```
Clearly should go in Devices, but which one? miscellaneous.py? may be better in aps_source.py? 

Then in code do 
```
from xxx import a_shutter_autoopen
 ...
yield from bps.mv(a_shutter_autoopen, 1)
```

prjemian, 2021-09-24:
Those are the right parts. Clearly not in apa stuff, look at the pv
prefix.  Best place is in devices/shutters.py

Then the action code goes in the before... plan.

jilavsky, 2021-09-24:
shutters.py is the right place for it. 
HELP: I would like to make it not talk to the device when USAXS is not in operations, in the "else:" case we have there... How do I simulate the PV when I do not want to talk to it? For shutters we have SimulatedApsPssShutterWithStatus, is there some simulated PV equivalent? 
I created branch 517_Enable_Autoopen_on_Collection.

jilavsky, 2021-09-24:
Added that to before_command_list, looks lot more logical there.

jilavsky, 2021-09-24:
How do I fake that PV when not in operations? Is there simulated PV?

prjemian, 2021-09-24:
When not in operations, you can simulate the `a_shutter_autoopen` with `ophyd.Signal` (no EPICS involved).  This will give you a signal to operate by `bps.mv`.  Next, you'll also want a simulated shutter (you probably already have one).

jilavsky, 2021-09-24:
Great, I just found it in suspenders.py, I think I got it in. Not sure why I need simulated shutter. You can check what I did in the branch when you get chance.

prjemian, 2021-09-24:
Maybe you don't need simulated A shutter.  You only need it if you plan to operate it from BS.

jilavsky, 2021-09-24:
We do not need to operate A shutter from BS, this is epics which will do it for us. This will reopen A shutter when allowed. We are trying to open shutter after beam dump. Last night staff forgot to check epics autoopen was enabled and after 2am beam dump the shutter did not reopen and when I can at 8am, BS was waiting for shutter to reopen. This is to prevent users from forgetting and losing the beamtime.

prjemian, 2021-09-24:
Then the easy simulator is `ophyd.Signal`

prjemian, 2021-09-24:
You have not pushed your branch to GH.  Must login to look at your branch.

prjemian, 2021-09-24:
Not sure why you need `_simulated_beam_in_hutch` and a suspender for it.  Why create a suspender for this anyway?

PR #518 (2021-09-24): Enable autoopen on collection
* fix #517
closed 2021-09-24 by prjemian