APS-USAXS/usaxs-bluesky-ended-2023 #519: add PTC10 heater

prjemian, 2021-09-27:
[Yesterday 5:22 PM] Ilavsky, Jan

> ... add PTC10 device (heater) and make that as one of the options in this heater program. ... Should be heater with same commands as Linkams...

prjemian, 2021-09-27:
Support for PTC10 is an issue (https://github.com/BCDA-APS/apstools/issues/491) in the milestone for release in `apstools` due this week.

prjemian, 2021-09-27:
from Jan:

> ... We use PTC10 routinely (like next few days) and it is supported in spec. I want to have basic stuff covered for bluesky...  We use number of heaters with it,. I can test it for you On Thursday or Friday when there is no beam.

prjemian, 2021-09-27:
GUI used by USAXS
[image]

prjemian, 2021-09-27:
This device might benefit from the [`apstools.devices.PVPositionerSoftDone`](https://github.com/BCDA-APS/apstools/pull/531), currently in review as a pull request.  The `PVPositionerSoftDone` device provides a software *done* signal that compares readback and setpoint to see if a move is approximately complete (within a tolerance).  This device will be useful for temperature controllers that do not provide an *at temperature* signal.

BUT, where does the user provide a set point in terms of temperature?

prjemian, 2021-09-27:
from Jan:

> Some of the PV names are not intuitive.  "High limit" is really target temperature,. This is ugly and complicated device and naming matches that.

prjemian, 2021-09-27:
I will refactor this from `ophyd.Device` to `ophyd.PVPositioner`.  For an example, see how it is used in the [temperature controller simulator](https://github.com/BCDA-APS/bluesky_training/blob/0fb04fa68508f9bb9014cb6dfbb3da2d90d62264/bluesky/instrument/devices/temperature_signal.py#L26-L41) in the APS bluesky instrument training.

prjemian, 2021-09-27:
Also will refactor RTD and AIO parts to be compatible with [plans in apstools](https://github.com/BCDA-APS/apstools/issues/491#issuecomment-928061296).

prjemian, 2021-09-27:
Here's what the `ptc10` object shows (both summary and EPICS values):
(Could change `aio` to `pid` for more meaningful name)

```python
In [54]: ptc10.summary()
data keys (* hints)
-------------------
*ptc10
 ptc10_coldj2
*ptc10_setpoint
*ptc10_temperatureB
 ptc10_temperatureC

read attrs
----------
readback             EpicsSignalRO       ('ptc10')
setpoint             EpicsSignalWithRBV  ('ptc10_setpoint')
temperatureB         EpicsSignalRO       ('ptc10_temperatureB')
temperatureC         EpicsSignalRO       ('ptc10_temperatureC')
coldj2               EpicsSignalRO       ('ptc10_coldj2')
rtd                  PTC10RtdChannel     ('ptc10_rtd')
aio                  PTC10AioChannel     ('ptc10_aio')

config keys
-----------
ptc10_aio_D
ptc10_aio_I
ptc10_aio_P
ptc10_aio_inputchoice
ptc10_aio_iotype
ptc10_aio_lowlimit
ptc10_aio_pidmode
ptc10_aio_ramprate
ptc10_aio_setpoint
ptc10_aio_tunelag
ptc10_aio_tunemode
ptc10_aio_tunestep
ptc10_aio_tunetype
ptc10_aio_voltage
ptc10_enable
ptc10_rtd_channelrange
ptc10_rtd_current
ptc10_rtd_power
ptc10_rtd_sensor
ptc10_rtd_units
ptc10_tolerance

configuration attrs
-------------------
tolerance            Signal              ('ptc10_tolerance')
enable               EpicsSignalWithRBV  ('ptc10_enable')
rtd                  PTC10RtdChannel     ('ptc10_rtd')
rtd.units            EpicsSignalRO       ('ptc10_rtd_units')
rtd.sensor           EpicsSignalWithRBV  ('ptc10_rtd_sensor')
rtd.channelrange     EpicsSignalWithRBV  ('ptc10_rtd_channelrange')
rtd.current          EpicsSignalWithRBV  ('ptc10_rtd_current')
rtd.power            EpicsSignalWithRBV  ('ptc10_rtd_power')
aio                  PTC10AioChannel     ('ptc10_aio')
aio.voltage          EpicsSignalRO       ('ptc10_aio_voltage')
aio.lowlimit         EpicsSignalWithRBV  ('ptc10_aio_lowlimit')
aio.iotype           EpicsSignalWithRBV  ('ptc10_aio_iotype')
aio.setpoint         EpicsSignalWithRBV  ('ptc10_aio_setpoint')
aio.ramprate         EpicsSignalWithRBV  ('ptc10_aio_ramprate')
aio.pidmode          EpicsSignalWithRBV  ('ptc10_aio_pidmode')
aio.P                EpicsSignalWithRBV  ('ptc10_aio_P')
aio.I                EpicsSignalWithRBV  ('ptc10_aio_I')
aio.D                EpicsSignalWithRBV  ('ptc10_aio_D')
aio.inputchoice      EpicsSignalWithRBV  ('ptc10_aio_inputchoice')
aio.tunelag          EpicsSignalWithRBV  ('ptc10_aio_tunelag')
aio.tunestep         EpicsSignalWithRBV  ('ptc10_aio_tunestep')
aio.tunemode         EpicsSignalWithRBV  ('ptc10_aio_tunemode')
aio.tunetype         EpicsSignalWithRBV  ('ptc10_aio_tunetype')

unused attrs
------------
done                 Signal              ('ptc10_done')
report_dmov_changes  Signal              ('ptc10_report_dmov_changes')

In [52]: %run -i ptc10_controller.py

In [53]: object_explorer(ptc10)
================ =============================== ========
name             PV reference                    value   
================ =============================== ========
aio.D            9idcTEMP:tc1:5A:pid:D_RBV       10.0    
aio.I            9idcTEMP:tc1:5A:pid:I_RBV       0.01    
aio.P            9idcTEMP:tc1:5A:pid:P_RBV       0.5     
aio.inputchoice  9idcTEMP:tc1:5A:pid:input_RBV   2A      
aio.iotype       9idcTEMP:tc1:5A:ioType_RBV      Meas out
aio.lowlimit     9idcTEMP:tc1:5A:lowLimit_RBV    0.0     
aio.pidmode      9idcTEMP:tc1:5A:pid:mode_RBV    Off     
aio.ramprate     9idcTEMP:tc1:5A:rampRate_RBV    1.0     
aio.setpoint     9idcTEMP:tc1:5A:setPoint_RBV    20.0    
aio.tunelag      9idcTEMP:tc1:5A:tune:lag_RBV    30.0    
aio.tunemode     9idcTEMP:tc1:5A:tune:mode_RBV   Off     
aio.tunestep     9idcTEMP:tc1:5A:tune:step_RBV   0.5     
aio.tunetype     9idcTEMP:tc1:5A:tune:type_RBV   Cons    
aio.voltage      9idcTEMP:tc1:5A:voltage_RBV     0.007   
coldj2           9idcTEMP:tc1:ColdJ2:temperature 0.0     
enable           9idcTEMP:tc1:outputEnable_RBV   0       
readback         9idcTEMP:tc1:2A:temperature     17.69   
rtd.channelrange 9idcTEMP:tc1:3A:range_RBV       Auto    
rtd.current      9idcTEMP:tc1:3A:current_RBV     Forward 
rtd.power        9idcTEMP:tc1:3A:power_RBV       Auto    
rtd.sensor       9idcTEMP:tc1:3A:sensor_RBV      Diode   
rtd.units        9idcTEMP:tc1:3A:units_RBV       Ohm     
setpoint         9idcTEMP:tc1:5A:highLimit_RBV   1200.0  
temperatureB     9idcTEMP:tc1:2B:temperature     17.393  
temperatureC     9idcTEMP:tc1:2C:temperature     0.0     
================ =============================== ========

```

prjemian, 2021-09-27:
```python
In [55]: ptc10.read()
Out[55]: 
OrderedDict([('ptc10', {'value': 17.664, 'timestamp': 1632769269.70712}),
             ('ptc10_setpoint',
              {'value': 1200.0, 'timestamp': 1632768657.286077}),
             ('ptc10_temperatureB',
              {'value': 17.349, 'timestamp': 1632769269.754152}),
             ('ptc10_temperatureC', {'value': 0.0, 'timestamp': 631152000.0}),
             ('ptc10_coldj2', {'value': 0.0, 'timestamp': 631152000.0})])
```

prjemian, 2021-09-27:
reformatting that:

```python
In [59]: listdevice(ptc10)
================== ====== ==========================
name               value  timestamp                 
================== ====== ==========================
ptc10              17.648 2021-09-27 14:02:14.701061
ptc10_setpoint     1200.0 2021-09-27 13:50:57.286077
ptc10_temperatureB 17.358 2021-09-27 14:02:14.849055
ptc10_temperatureC 0.0    1989-12-31 18:00:00       
ptc10_coldj2       0.0    1989-12-31 18:00:00       
================== ====== ==========================
```

prjemian, 2021-09-27:
@jilavsky Except for possible name changes, this is ready for testing later this week.

prjemian, 2021-09-27:
```
In [60]: ptc10.position
Out[60]: 17.664
```

prjemian, 2021-09-27:
now

```python
In [62]: object_explorer(ptc10)
================ =============================== ========
name             PV reference                    value   
================ =============================== ========
coldj2           9idcTEMP:tc1:ColdJ2:temperature 0.0     
enable           9idcTEMP:tc1:outputEnable_RBV   0       
pid.D            9idcTEMP:tc1:5A:pid:D_RBV       10.0    
pid.I            9idcTEMP:tc1:5A:pid:I_RBV       0.01    
pid.P            9idcTEMP:tc1:5A:pid:P_RBV       0.5     
pid.inputchoice  9idcTEMP:tc1:5A:pid:input_RBV   2A      
pid.iotype       9idcTEMP:tc1:5A:ioType_RBV      Meas out
pid.lowlimit     9idcTEMP:tc1:5A:lowLimit_RBV    0.0     
pid.pidmode      9idcTEMP:tc1:5A:pid:mode_RBV    Off     
pid.ramprate     9idcTEMP:tc1:5A:rampRate_RBV    1.0     
pid.setpoint     9idcTEMP:tc1:5A:setPoint_RBV    20.0    
pid.tunelag      9idcTEMP:tc1:5A:tune:lag_RBV    30.0    
pid.tunemode     9idcTEMP:tc1:5A:tune:mode_RBV   Off     
pid.tunestep     9idcTEMP:tc1:5A:tune:step_RBV   0.5     
pid.tunetype     9idcTEMP:tc1:5A:tune:type_RBV   Cons    
pid.voltage      9idcTEMP:tc1:5A:voltage_RBV     0.007   
readback         9idcTEMP:tc1:2A:temperature     17.621  
rtd.channelrange 9idcTEMP:tc1:3A:range_RBV       Auto    
rtd.current      9idcTEMP:tc1:3A:current_RBV     Forward 
rtd.power        9idcTEMP:tc1:3A:power_RBV       Auto    
rtd.sensor       9idcTEMP:tc1:3A:sensor_RBV      Diode   
rtd.units        9idcTEMP:tc1:3A:units_RBV       Ohm     
setpoint         9idcTEMP:tc1:5A:highLimit_RBV   1200.0  
temperatureB     9idcTEMP:tc1:2B:temperature     17.346  
temperatureC     9idcTEMP:tc1:2C:temperature     0.0     
================ =============================== ========
```

prjemian, 2021-09-29:
Suggest adding a STOP feature per https://github.com/BCDA-APS/bluesky_training/issues/35

Add a STOP feature so the controller will hold the current readback temperature when the `stop()` method is called.

jilavsky, 2022-01-12:
This will happen with installation of new release (1.6.0rc5+) of apstools.

prjemian, 2022-01-23:
Follow the [example](https://github.com/BCDA-APS/apstools/blob/a4531af1ad4ee0d4f7cd1cc96f8640edc3b0efc7/apstools/devices/ptc10_controller.py#L28-L39):

```python
from apstools.devices import PTC10PositionerMixin
from ophyd import PVPositioner

class MyPTC10(PTC10PositionerMixin, PVPositioner):
      readback = Component(EpicsSignalRO, "2A:temperature", kind="hinted")
      setpoint = Component(EpicsSignalWithRBV, "5A:setPoint", kind="hinted")
      rtd = Component(PTC10RtdChannel, "3A:")
      pid = Component(PTC10AioChannel, "5A:")

ptc10 = MyPTC10("9idcTEMP:tc1:", name="ptc10")
ptc10.report_dmov_changes.put(True)  # a diagnostic
ptc10.tolerance.put(1.0)  # done when |readback-setpoint|<=tolerance
```

commit ab4c0fc8 (2022-01-23): FIX #519
closed 2022-01-23 by prjemian