APS-USAXS/usaxs-bluesky-ended-2023 #520: Set feedback on when pausing for beam dump

jilavsky, 2021-10-09:
Beam dump happened at time, when feedback was not set to on in program. 100s delay after beam comes back is more useful, it feedback is on. We need to set feedback on when pausing and after 100sec set feedback to what it was before pausing.

jilavsky, 2021-11-26:
Coming back to this. This seems above my abilities. Reading  suspenders.py we are simply installing generic suspenders from BS main under specific conditions and use either of those. The trouble is, that I think I need this one:
```
 suspender_white_beam_ready = bluesky.suspenders.SuspendBoolLow(
        white_beam_ready.available, sleep=100
    )
    RE.install_suspender(suspender_white_beam_ready)
```
to do one more activity, when engaged. Not simply suspend only, but switch on the feedback 
We need somewhere (following scans.py) do:
```
from ..devices import monochromator, MONO_FEEDBACK_OFF, MONO_FEEDBACK_ON
and 
yield from bps.mv(
        monochromator.feedback.on, MONO_FEEDBACK_ON,
        timeout=MASTER_TIMEOUT,
        )
```
This needs to be done anytime before 100sec recovery of beam starts. Unluckily, modifying bluesky.suspenders.SuspendBoolLow is not in my abilities. If I understand correctly, we may need to copy the default one as our own and add this feature? Guidance needed...

prjemian, 2022-01-28:
You've described the approach I'd take.  Make a local subclass of `bluesky.suspenders.SuspendBoolLow()` and add the features you described.  I'll take a look.

prjemian, 2022-01-28:
Looking at the source code for `SuspendBoolLow()`, it (and the other standard suspenders) has support for addition keyword arguments that can supply a plan to execute `pre` and `post`:

```python
    pre_plan : iterable or iterator, optional
            a generator, list, or similar containing `Msg` objects
    post_plan : iterable or iterator, optional
            a generator, list, or similar containing `Msg` objects
```

Later in the [source code](https://github.com/bluesky/bluesky/blob/05238c2177c0d4b5e58905d65395b079454aabfb/bluesky/suspenders.py#L610-L616), there are these definitions:

```text
    pre_plan : iterable or callable, optional
       Plan to execute just before suspending. If callable, must
       take no arguments.
    post_plan : iterable or callable, optional
        Plan to execute just before resuming. If callable, must
        take no arguments.
```

prjemian, 2022-01-28:
Which means, we (you?) write two simple plans that do the additional things you want.  Then we change https://github.com/APS-USAXS/ipython-usaxs/blob/7affdc1c9cd7f4738112d7e6c068fbc30dad9768/profile_bluesky/startup/instrument/devices/suspenders.py#L37-L40 at line 38 to read

```python
    white_beam_ready.available, sleep=100,  # <-- need this trailing comma
    pre_plan=mono_beam_lost_plan,
    post_plan=mono_beam_just_came_back_plan,
```

These are the two plans:

```python
previous_feedback = None  # to remember the feedback setting during a dump

def mono_beam_lost_plan():
    global previous_feedback  # needed if we change this value

    previous_feedback  = monochromator.feedback.on.get()
    yield from bps.mv(
        monochromator.feedback.on, MONO_FEEDBACK_ON,
        timeout=MASTER_TIMEOUT,
    )

def mono_beam_just_came_back_plan():
    yield from bps.mv(
        monochromator.feedback.on, previous_feedback,
        timeout=MASTER_TIMEOUT,
    )
```

All of this goes into the `suspenders.py` file.

BUT, you want that feedback on during the 100s delay.  Need to explore that one.  Does it sleep then call post_plan?  Or the other way around?  Handled by [`RE.request_suspend()`](https://github.com/bluesky/bluesky/blob/05238c2177c0d4b5e58905d65395b079454aabfb/bluesky/run_engine.py#L1137)

prjemian, 2022-01-29:
Tested locally and confirmed that `post_plan` is called _after_ the `sleep` delay.  These features will do the tasks you want!

prjemian, 2022-01-29:
There is a `MASTER_TIMEOUT` : https://github.com/APS-USAXS/ipython-usaxs/blob/7affdc1c9cd7f4738112d7e6c068fbc30dad9768/profile_bluesky/startup/instrument/plans/scans.py#L74

If we use that here, we need to move where it is defined (so we do not have a wicked infinite loop with `import` statements).

prjemian, 2022-01-29:
`MASTER_TIMEOUT` will not be used here.  Hope the local definition here of a timeout value is never a problem.

prjemian, 2022-01-29:
Enabled in commit 30b1e72

PR #542 (2022-01-30): Ensure feedback is ON during delay period when resuming from beam dump
- FIX #520
closed 2022-01-30 by jilavsky