APS-USAXS/usaxs-bluesky-ended-2023 #521: When using Linkam plans decorations are not done

jilavsky, 2021-10-10:
Using RE(myLinkamPlan...) doe snot set PV for Collecting data and does not run start of data collection (dark currents for detectors etc.)

jilavsky, 2021-10-10:
Need to import proper command and add this to the list at the beginning of main body

jilavsky, 2021-10-10:
from instrument.plans import before_command_list 
should work?

jilavsky, 2021-10-11:
OK, I will change that tomorrow and try it next weekend.

jilavsky, 2021-10-13:
I have modified current linkam_local.py on usaxscontrol to have:

```python
yield from before_command_list() before se start time t0
```
and
```python
yield from  after_command_list()  after all is done. 
```

jilavsky, 2021-10-18:
Tried to use this, failed here:
  yield from before_command_list()            #this will run usual startup scripts for scans
on
      verify_commands(commands)
I passes nothing in this, assuming commands=None is good enough to pass through verify_commands
it failed... 

```python
In [9]: RE(myLinkamPlan(0, 0, 0.103, "AB174", 650, 200, 3600, 40, 1))
   ...: 
I Sun-19:50:23 - Linkam controller PV prefix=9idcLINKAM:tc1:
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-9-2de22bc582dd> in <module>
----> 1 RE(myLinkamPlan(0, 0, 0.103, "AB174", 650, 200, 3600, 40, 1))

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    805             self._task_fut.add_done_callback(set_blocking_event)
    806 
--> 807         self._resume_task(init_func=_build_task)
    808 
    809         if self._interrupted:

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in _resume_task(self, init_func)
    929                     if (exc is not None
    930                             and not isinstance(exc, _RunEnginePanic)):
--> 931                         raise exc
    932 
    933     def install_suspender(self, suspender):

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1498             exit_reason = str(err)
   1499             self.log.exception("Run aborted")
-> 1500             raise err
   1501         finally:
   1502             if not exit_reason:

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1363                     else:
   1364                         try:
-> 1365                             msg = self._plan_stack[-1].send(resp)
   1366                         # We have exhausted the top generator
   1367                         except StopIteration:

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in __call__(self, plan)
   1305         plan = monitor_during_wrapper(plan, self.monitors)
   1306         plan = baseline_wrapper(plan, self.baseline)
-> 1307         return (yield from plan)
   1308 
   1309 

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in baseline_wrapper(plan, devices, name)
   1160         return (yield from plan)
   1161     else:
-> 1162         return (yield from plan_mutator(plan, insert_baseline))
   1163 
   1164 

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
    168                     continue
    169                 else:
--> 170                     raise ex
    171         # if inserting / mutating, put new generator on the stack
    172         # and replace the current msg with the first element from the

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
    121             ret = result_stack.pop()
    122             try:
--> 123                 msg = plan_stack[-1].send(ret)
    124             except StopIteration as e:
[... 70 lines cut ...]
    125                 # discard the exhausted generator

~/.ipython/user/linkam_local.py in myLinkamPlan(pos_X, pos_Y, thickness, scan_title, temp1, rate1, delay1, temp2, rate2, md)
     48     logger.info(f"Linkam controller PV prefix={linkam.prefix}")
     49 
---> 50     yield from before_command_list()            #this will run usual startup scripts for scans
     51 
     52     t0 = time.time()

~/.ipython/profile_bluesky/startup/instrument/plans/command_list.py in before_command_list(md, commands)
    103     from .scans import preUSAXStune
    104 
--> 105     verify_commands(commands)
    106 
    107     if md is None:

~/.ipython/profile_bluesky/startup/instrument/plans/command_list.py in verify_commands(commands)
    154     # separate commands into individual components, see execute_command_list for details
    155     scan_actions = "flyscan usaxsscan saxs saxsexp waxs waxsexp".split()
--> 156     for command in commands:
    157         action, args, i, raw_command = command
    158         if action.lower() in scan_actions:

TypeError: 'NoneType' object is not iterable
```

jilavsky, 2021-10-18:
May be defining my own before_command_list in that function is the right thing to do?

jilavsky, 2021-11-12:
@prjemian created Branch Prevent-before_command_list-from-failing-in-Linkam-code-#512. Can you look at it? 
there is single line change which prevents verify_commands if commands are None. 

I think this is trivial fix for my Linkam issue.

prjemian, 2022-01-12:
Fix applied here: https://github.com/APS-USAXS/ipython-usaxs/blob/a9357a7755d7c37c54f419c8b62c42b1eefe8a15/profile_bluesky/startup/instrument/plans/command_list.py#L105-L106

closed 2022-01-12 by jilavsky