APS-USAXS/usaxs-bluesky-ended-2023 #525: Create code to repeat usaxs.mac

jilavsky, 2021-11-27:
I will need explanation and help with code to run usaxs.mac multiple times (in a loop). I got myself into pickle with making unstoppable code by looping run_command_file and assuming I can abort from it using our abort PV. It is not working and took down BS really badly.

code is here: https://github.com/APS-USAXS/ipython-usaxs/blob/master/user/cycleUsaxsMacFile.py#:~:text=def%20cycle_command_file(filename,Status

It did not stop on PV with abort (I was afraid of that ahead) and repeated ctrl-C resulted in:
```
In [8]: RE.abort()
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-8-7af2a2d1869f> in <module>
----> 1 RE.abort()

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in abort(self, reason)
   1112         :meth:`RunEngine.stop`
   1113         """
-> 1114         return self.__interrupter_helper(self._abort_coro(reason))
   1115 
   1116     async def _abort_coro(self, reason):

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in __interrupter_helper(self, coro)
   1177     def __interrupter_helper(self, coro):
   1178         if self.state == 'panicked':
-> 1179             raise RuntimeError("The RunEngine is panicked and "
   1180                                "cannot be recovered. "
   1181                                "You must restart bluesky.")

RuntimeError: The RunEngine is panicked and cannot be recovered. You must restart bluesky.
```
Had to quit BS and even that did not help, could not get back command prompt, had to actually kill terminal. BS is really persistent, but too much for its own good - or for this user good...

Anyway @prjemian, we need to talk when convenient and I need to crack this. Clearly, my dummy approach did not work ;-) This is no rush at all, I just thought I can handle this on my own.

prjemian, 2021-11-27:
> Assigned #525 <https://github.com/APS-USAXS/ipython-usaxs/issues/525> to

At least, this time, we were not bombed with 2,714 email messages.

jilavsky, 2022-01-13:
This may not be necessary, prefer to export multiple copies of commands, which is now implemented in Igor support code. Closing for now.

closed 2022-01-13 by jilavsky