APS-USAXS/usaxs-bluesky-ended-2023 #526: BS crashed mid way of heater run

jilavsky, 2021-11-28:
User was running 12+ hours annealing run using Linkam heater, BS crashed mid way (2:54am) ruining the sample and loosing 12+ hours of beamtime. It crashed on limit error while changing geometry, something done 1000x before. 
This was first user I pushed to run in BS and I tested this in October. We just lost trust of expert user who could have helped to advertise BS. 

```
I Sun-02:54:51 - Removed SAXS from beam position
I Sun-02:54:51 - Moving USAXS out of beam
I Sun-02:54:51 - Removed USAXS from beam position
---------------------------------------------------------------------------
LimitError                                Traceback (most recent call last)
<ipython-input-71-483ccfbab2ee> in <module>
----> 1 RE(myLinkamPlan(0,0,0.109,"AR174AB",483,200,660,40,200))

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in __call__(self, *args, **metadata_kw)
    805             self._task_fut.add_done_callback(set_blocking_event)
    806 
--> 807         self._resume_task(init_func=_build_task)
    808 
    809         if self._interrupted:

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in _resume_task(self, init_func)
    929                     if (exc is not None
    930                             and not isinstance(exc, _RunEnginePanic)):
--> 931                         raise exc
    932 
    933     def install_suspender(self, suspender):

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1498             exit_reason = str(err)
   1499             self.log.exception("Run aborted")
-> 1500             raise err
   1501         finally:
   1502             if not exit_reason:

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/run_engine.py in _run(self)
   1339                         # throw the exception at the current plan
   1340                         try:
-> 1341                             msg = self._plan_stack[-1].throw(
   1342                                 stashed_exception or resp)
   1343                         except Exception as e:

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in __call__(self, plan)
   1305         plan = monitor_during_wrapper(plan, self.monitors)
   1306         plan = baseline_wrapper(plan, self.baseline)
-> 1307         return (yield from plan)
   1308 
   1309 

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in baseline_wrapper(plan, devices, name)
   1160         return (yield from plan)
   1161     else:
-> 1162         return (yield from plan_mutator(plan, insert_baseline))
   1163 
   1164 

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
     76             # if we have a stashed exception, pass it along
     77             try:
---> 78                 msg = plan_stack[-1].throw(exception)
     79             except StopIteration as e:
     80                 # discard the exhausted generator

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in monitor_during_wrapper(plan, signals)
    801     plan1 = plan_mutator(plan, insert_after_open)
    802     plan2 = plan_mutator(plan1, insert_before_close)
--> 803     return (yield from plan2)
    804 
    805 

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/bluesky/preprocessors.py in plan_mutator(plan, msg_proc)
     76             # if we have a stashed exception, pass it along
[... 143 lines cut ...]
    171         try:

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/ophyd/positioner.py in move(self, position, moved_cb, timeout)
    189             timeout = self._timeout
    190 
--> 191         self.check_value(position)
    192 
    193         self._run_subs(sub_type=self._SUB_REQ_DONE, success=False)

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/ophyd/epics_motor.py in check_value(self, pos)
    234     def check_value(self, pos):
    235         '''Check that the position is within the soft limits'''
--> 236         self.user_setpoint.check_value(pos)
    237 
    238     @required_for_connection

/share1/bluesky/conda_envs/bluesky_2021_1/lib/python3.8/site-packages/ophyd/signal.py in check_value(self, value)
   1563 
   1564         if not (low_limit <= value <= high_limit):
-> 1565             raise LimitError('Value {} outside of range: [{}, {}]'
   1566                              .format(value, low_limit, high_limit))
   1567 

LimitError: Value 261.0 outside of range: [-5.0, 5.0]
```

prjemian, 2022-01-12:
We expect that #528 will avoid this situation in the future.

prjemian, 2022-01-12:
Closing this with expectation that #528 will avoid this situation in the future.