APS-USAXS/usaxs-bluesky-ended-2023 #529: Is this correct?

jilavsky, 2021-12-21:
APS cycle ended. BS was running and I was expecting simple white beam suspension. Our flyscan seem to have somehow finished? Is this expected behavior? Just checking

```
Transient Scan ID: 1028     Time: 2021-12-21 07:59:56
Persistent Unique Scan ID: '46f93e69-e065-4665-8a96-7ed817ad5432'
New stream: 'baseline'
New stream: 'aps_current_monitor'
I Tue-07:59:57 - HDF5 config: /share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml
I Tue-07:59:57 - HDF5 output: /share1/USAXS_data/2021-12/12_20_HuPlate2/12_20_HuPlate2_usaxs/WBG__L_0308.h5
D Tue-07:59:57 - progress_reporting has arrived
I Tue-07:59:57 - flying, s    ar, deg      ay, mm       dy, mm       channel      elapsed, s 
D Tue-08:00:02 - 5.01         8.7420736    0.00681      13.40471     0            0.00       
Suspending....To get prompt hit Ctrl-C twice to pause.
Suspension occurred at 2021-12-21 08:00:05.
Justification for this suspension:
Signal usaxs_CheckBeamStandard is low
Suspending....To get prompt hit Ctrl-C twice to pause.
Suspension occurred at 2021-12-21 08:00:05.
Justification for this suspension:
Signal white_beam_ready_available is low
D Tue-08:00:07 - 10.01        8.7413942    0.00435      13.39363     14           3.88       
D Tue-08:00:12 - 15.01        8.7413942    0.00435      13.39363     14           8.88       
D Tue-08:00:17 - 20.01        8.7403870    0.00027      13.37621     66           13.88      
D Tue-08:00:22 - 25.02        8.7397987    -0.00122     13.32685     81           18.90      
D Tue-08:00:27 - 30.03        8.7405523    0.00081      13.16417     124          23.90      
D Tue-08:00:32 - 35.03        8.7398006    -0.00120     12.85742     143          28.92      
D Tue-08:00:37 - 40.03        8.7407804    0.00146      12.40148     179          33.92      
D Tue-08:00:42 - 45.03        8.7398087    -0.00119     11.78502     205          38.92      
D Tue-08:00:47 - 50.04        8.7410158    0.00243      10.98255     234          43.92      
D Tue-08:00:52 - 55.04        8.7398024    -0.00116     10.01069     268          48.92      
D Tue-08:00:57 - 60.04        8.7413394    0.00319      8.87499      286          53.92      
D Tue-08:01:02 - 65.05        8.7397889    -0.00122     7.51538      331          58.93      
D Tue-08:01:07 - 70.05        8.7417451    0.00418      5.97807      334          63.93      
D Tue-08:01:12 - 75.05        8.7397953    -0.00123     4.21052      394          68.93      
D Tue-08:01:17 - 80.06        8.7406911    0.00559      2.28749      394          73.93      
D Tue-08:01:22 - 85.06        8.7398054    -0.00121     0.09172      456          78.93      
D Tue-08:01:27 - 90.07        8.7398054    0.00027      -2.24429     456          83.95      
D Tue-08:01:32 - 95.08        8.7398064    -0.00115     -4.87293     519          88.95      
D Tue-08:01:37 - 100.08       8.7398035    -0.00115     -7.66558     519          93.97      
D Tue-08:01:42 - 105.09       8.7398996    -0.00116     -10.67718    581          98.97      
D Tue-08:01:47 - 110.09       8.7398215    -0.00116     -12.88761    581          103.97     
D Tue-08:01:52 - 115.10       8.7400103    -0.00116     -12.88761    641          108.97     
D Tue-08:01:57 - 120.11       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:02:02 - 125.11       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:02:07 - 130.11       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:02:12 - 135.11       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:02:17 - 140.12       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:02:22 - 145.13       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:02:27 - 150.13       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:02:32 - 155.13       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:02:37 - 160.13       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:02:42 - 165.14       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:02:47 - 170.14       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:02:52 - 175.15       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:02:57 - 180.16       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:03:02 - 185.16       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:03:07 - 190.17       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:03:12 - 195.18       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:03:17 - 200.18       8.7397967    -0.00116     -12.88761    647          110.54     
D Tue-08:03:22 - 205.18       8.7397967    -0.00116     -12.88761    647          110.54     
I Tue-08:03:27 - 210.01       8.7397967    -0.00116     -12.88761    647          110.54     
E Tue-08:03:27 - 210.0103838443756s - progress_reporting timeout!!
```

prjemian, 2021-12-21:
Yes.  Suspenders work on a finer _granularity_ than a scan.  As the RunEngine processes a [`Msg`](https://github.com/bluesky/bluesky/blob/dc9819b7b903e3950cb79d0e473eb67c9c126327/bluesky/utils/__init__.py#L39) object, it checks all suspnders, amongst its other background activities.  A `Msg()` is the core action of a plan which is handled by the RunEngine.  Here's a list (it is possible to change this list):

```python
In [1]: RE.commands
Out[1]: 
['create',
 'save',
 'drop',
 'read',
 'monitor',
 'unmonitor',
 'null',
 'RE_class',
 'stop',
 'set',
 'trigger',
 'sleep',
 'wait',
 'checkpoint',
 'clear_checkpoint',
 'rewindable',
 'pause',
 'resume',
 'collect',
 'kickoff',
 'complete',
 'configure',
 'stage',
 'unstage',
 'subscribe',
 'unsubscribe',
 'open_run',
 'close_run',
 'wait_for',
 'input',
 'install_suspender',
 'remove_suspender']
```

prjemian, 2021-12-21:
Which means the scan will not complete normally.

closed 2021-12-21 by prjemian

prjemian, 2021-12-21:
It is possible to see the sequence of `Msg()` objects from any plan using `bluesky.simulators.summarize_plan()` instead of the `RE()`.  Here is an example:

```python
In [12]: import bluesky.simulators

In [13]: bluesky.simulators.summarize_plan(bp.rel_scan([noisy], m1, -1, 1, 5))
=================================== Open Run ===================================
m1 -> -1.0
  Read ['noisy', 'm1']
m1 -> -0.5
  Read ['noisy', 'm1']
m1 -> 0.0
  Read ['noisy', 'm1']
m1 -> 0.5
  Read ['noisy', 'm1']
m1 -> 1.0
  Read ['noisy', 'm1']
================================== Close Run ===================================
m1 -> 0.0
```