APS-USAXS/usaxs-bluesky-ended-2023 #532: BlueSky and "osciallte.py" ?

jilavsky, 2022-01-14:
We have little piece of python code ~/bin/oscillate.py. When run it oscillates sy between two points to average larger area during the SAXS/WAXS data collection. Typical use is to average over USAXS area during SAXS/WAXS. 
We have a bit on LAX which is used to start/stop. We start before SAXS data exposure and stop after and same for WAXS. 
spec does not care that something else is moving motor sy while collecting data on AD.
What will BlueSky do? Can we use the same method or how do we engineer this functionality?  This is not urgent, but would be useful. We are talking about using oscillation during SAXS/WAXS data collection by 0.3mm up and 0.3mm down as default for users in the future.

prjemian, 2022-01-14:
Can use the same functionality.  Having been the one who probably engineered this activity, I still forget.

As long as BS is not moving the motor, it, too, will not care if the motor is moved externally.

Since it starts with Py2.7, we should test it with Py3 (with the Bluesky environment), even though it uses pure PyEpics (no ophyd):

```python
#!/APSshare/anaconda/x86_64/bin/python

import epics
import time

motorName = '9idcLAX:m58:c2:m2'
controlName = '9idcLAX:bit3.VAL'

# ...
```

jilavsky, 2022-01-14:
This is Kevin's doing, if I remember. We will need to modify this code, but Ivan indicated that he is looking for small project like this. Let us wait if it gets anywhere in his hands. Closing for now.

closed 2022-01-14 by jilavsky