APS-USAXS/usaxs-bluesky-ended-2023 #536: PTC10 not connected today

prjemian, 2022-01-24:
On bluesky startup today, the PTC10 device cannot connect with EPICS:

```python
In [9]: ptc10 = USAXS_PTC10("9idcTEMP:tc1:", name="ptc10")
   ...: ptc10.report_dmov_changes.put(True)  # a diagnostic
   ...: ptc10.tolerance.put(1.0)  # done when |readback-setpoint|<=tolerance
   ...:

In [10]: ptc10.connected
Out[10]: False
```

The PTC10 is not used by existing code except on demand by a user, so there is no bug.  But the support is new and needs to be tested before a user requests to use it.

jilavsky, 2022-01-24:
When one of us goes to work, we will setup PTC 10 for your testing needs. Ivan is not sure when he gets there, but likely tomorrow. I'll send you message and leave it on for day or two as needed.

prjemian, 2022-01-25:
Works as expected:

```python
In [1]: from instrument.devices.ptc10_controller import ptc10
I Tue-16:34:30 - /home/beams/USAXS/.ipython/profile_bluesky/startup/instrument/devices/ptc10_controller.py

In [2]: ptc10.connected
Out[2]: True

In [3]: ptc10.read()
Out[3]: 
OrderedDict([('ptc10', {'value': 29.976, 'timestamp': 1643150081.199434}),
             ('ptc10_setpoint',
              {'value': 30.0, 'timestamp': 1643150070.6837475})])

In [4]: ptc10.move(50)
Out[4]: MoveStatus(done=True, pos=ptc10, elapsed=30.2, success=True, settle_time=0.0)

In [5]: ptc10.position
Out[5]: 51.03

In [6]: ptc10.move(30)
Out[6]: MoveStatus(done=True, pos=ptc10, elapsed=146.1, success=True, settle_time=0.0)
```

prjemian, 2022-01-25:
Looking a bit deeper:

```python
In [7]: listdevice(ptc10)
Out[7]: 
                    data name       value                  timestamp
0                  ptc10_done        True 2022-01-25 16:41:03.208333
1             ptc10_tolerance           1 2022-01-25 16:34:30.573542
2   ptc10_report_dmov_changes        True 2022-01-25 16:34:30.573523
3                       ptc10      29.772 2022-01-25 16:41:03.207392
4              ptc10_setpoint          30 2022-01-25 16:36:01.851343
5                ptc10_enable          On 2022-01-25 16:34:30.683871
6          ptc10_temperatureB         NaN 2022-01-25 16:41:03.255347
7          ptc10_temperatureC           0 1989-12-31 18:00:00.000000
8                ptc10_coldj2           0 1989-12-31 18:00:00.000000
9       ptc10_rtd_temperature         NaN 2022-01-25 16:41:03.353343
10            ptc10_rtd_units         Ohm 2022-01-25 14:10:48.412345
11           ptc10_rtd_sensor  Thermistor 2022-01-25 16:34:30.693836
12     ptc10_rtd_channelrange        Auto 2022-01-25 16:34:30.696734
13          ptc10_rtd_current     Forward 2022-01-25 16:34:30.700812
14            ptc10_rtd_power        Auto 2022-01-25 16:34:30.703655
15          ptc10_pid_voltage       1.876 2022-01-25 16:41:03.453284
16        ptc10_pid_highlimit         300 2022-01-25 16:34:30.708300
17         ptc10_pid_lowlimit           0 2022-01-25 16:34:30.711283
18           ptc10_pid_iotype    Meas out 2022-01-25 16:34:30.714294
19         ptc10_pid_setpoint          30 2022-01-25 16:36:01.851343
20         ptc10_pid_ramprate           1 2022-01-25 16:34:30.717524
21        ptc10_pid_offswitch           0 1989-12-31 18:00:00.000000
22          ptc10_pid_pidmode          On 2022-01-25 16:34:30.721967
23                ptc10_pid_P         0.5 2022-01-25 16:34:30.724475
24                ptc10_pid_I        0.03 2022-01-25 16:34:30.727206
25                ptc10_pid_D         1.7 2022-01-25 16:34:30.730006
26      ptc10_pid_inputchoice          2A 2022-01-25 16:34:30.733690
27          ptc10_pid_tunelag          30 2022-01-25 16:34:30.737138
28         ptc10_pid_tunestep         0.5 2022-01-25 16:34:30.739357
29         ptc10_pid_tunemode         Off 2022-01-25 16:34:30.742490
30         ptc10_pid_tunetype        Cons 2022-01-25 16:34:30.745599
```

these two signals are not updated from the hardware:

```
7          ptc10_temperatureC           0 1989-12-31 18:00:00.000000
8                ptc10_coldj2           0 1989-12-31 18:00:00.000000
```

and these two signals are NaN:

```
6          ptc10_temperatureB         NaN 2022-01-25 16:41:03.255347
9       ptc10_rtd_temperature         NaN 2022-01-25 16:41:03.353343
```

I will comment out all four of these for now.

prjemian, 2022-01-25:
Can't get rid of ptc10.rtd.temperature and keep the rtd, removing the rtd by comment.

commit d2dafc57 (2022-01-25): FIX #536 remove parts that have no hardware signal
closed 2022-01-25 by prjemian