APS-USAXS/usaxs-bluesky-ended-2023 #545: spec flyscan saving data

jilavsky, 2022-02-01:
This is not BlueSky, but spec flyscan saving data. Looks to me that spec code saving data is looking at old bss ioc PVs. Can you fix it, please? 

```
Tue Feb 01 11:53:01 2022.  Please wait approximately 105s to complete FlyScan..
Not connected PV=9id_bss:activity  ophyd=metadata_0331
Not connected PV=9id_bss:badge  ophyd=metadata_0332
Not connected PV=9id_bss:bss_name  ophyd=metadata_0333
Not connected PV=9id_bss:contact  ophyd=metadata_0334
Not connected PV=9id_bss:email  ophyd=metadata_0335
Not connected PV=9id_bss:institution  ophyd=metadata_0336
Not connected PV=9id_bss:station  ophyd=metadata_0337
Not connected PV=9id_bss:team_others  ophyd=metadata_0338
Not connected PV=9id_bss:time_begin  ophyd=metadata_0339
Not connected PV=9id_bss:time_end  ophyd=metadata_0340
Not connected PV=9id_bss:timestamp  ophyd=metadata_0341
Not connected PV=9id_bss:title  ophyd=metadata_0342
preliminaryWriteFile(): PV 9id_bss:activity is not connected now
preliminaryWriteFile(): PV 9id_bss:badge is not connected now
preliminaryWriteFile(): PV 9id_bss:bss_name is not connected now
preliminaryWriteFile(): PV 9id_bss:contact is not connected now
preliminaryWriteFile(): PV 9id_bss:email is not connected now
preliminaryWriteFile(): PV 9id_bss:institution is not connected now
preliminaryWriteFile(): PV 9id_bss:station is not connected now
preliminaryWriteFile(): PV 9id_bss:team_others is not connected now
preliminaryWriteFile(): PV 9id_bss:time_begin is not connected now
preliminaryWriteFile(): PV 9id_bss:time_end is not connected now
preliminaryWriteFile(): PV 9id_bss:timestamp is not connected now
preliminaryWriteFile(): PV 9id_bss:title is not connected now
saveFile(): PV 9id_bss:activity is not connected now
saveFile(): PV 9id_bss:badge is not connected now
saveFile(): PV 9id_bss:bss_name is not connected now
saveFile(): PV 9id_bss:contact is not connected now
saveFile(): PV 9id_bss:email is not connected now
saveFile(): PV 9id_bss:institution is not connected now
saveFile(): PV 9id_bss:station is not connected now
saveFile(): PV 9id_bss:team_others is not connected now
saveFile(): PV 9id_bss:time_begin is not connected now
saveFile(): PV 9id_bss:time_end is not connected now
saveFile(): PV 9id_bss:timestamp is not connected now
saveFile(): PV 9id_bss:title is not connected now
```

prjemian, 2022-02-01:
APS BSS IOC prefix | description
-- | ---
**`9idc:bss:`** | **use this one now**
`9id_bss:` | old, do not use now

prjemian, 2022-02-01:
Also:

```bash
usaxs_support/builder_382.py:    <Attribute name="BSS_activity" type="EPICS_PV" source="9id_bss:activity"       dbrtype="DBR_STRING"     des
usaxs_support/saveFlyData.xml:                                  <PV label="BSS_activity"                pvname="9id_bss:activity" string="tr
usaxs_support/saveFlyData.xml:                                  <PV label="BSS_badge"                   pvname="9id_bss:badge" string="true"
usaxs_support/saveFlyData.xml:                                  <PV label="BSS_bss_name"                pvname="9id_bss:bss_name" string="tr
usaxs_support/saveFlyData.xml:                                  <PV label="BSS_contact"                 pvname="9id_bss:contact" string="tru
usaxs_support/saveFlyData.xml:                                  <PV label="BSS_email"                   pvname="9id_bss:email" string="true"
usaxs_support/saveFlyData.xml:                                  <PV label="BSS_institution"     pvname="9id_bss:institution" string="true" /
usaxs_support/saveFlyData.xml:                                  <PV label="BSS_station"                 pvname="9id_bss:station" string="tru
usaxs_support/saveFlyData.xml:                                  <PV label="BSS_team_others"     pvname="9id_bss:team_others" string="true" /
usaxs_support/saveFlyData.xml:                                  <PV label="BSS_time_begin"              pvname="9id_bss:time_begin" string="
usaxs_support/saveFlyData.xml:                                  <PV label="BSS_time_end"                pvname="9id_bss:time_end" string="tr
usaxs_support/saveFlyData.xml:                                  <PV label="BSS_timestamp"               pvname="9id_bss:timestamp" string="t
usaxs_support/saveFlyData.xml:                                  <PV label="BSS_title"                   pvname="9id_bss:title" string="true"
```

prjemian, 2022-02-01:
But `usaxs_support/saveFlyData.xml` has this note:

```
!!!!!!!!! This is AN EXAMPLE file !!!!!!!
The real one is in the area detector configuration directory now
    /share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml
```

prjemian, 2022-02-01:
Interesting.  References to the `9id_bss:` PV prefix exist **only** in the _EXAMPLE_ file, not in the file we are advised to use.

use | file
--- | ---
operations | `/share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml`
example | `/home/beams/USAXS/.ipython/profile_bluesky/startup/usaxs_support/saveFlyData.xml`

* Which software is using the wrong file?
* Also, the example should be renamed but that might cause immediate problems for the current users.

#### search for `saveFlyData.xml`

```bash
bash-4.2$ git grep saveFlyData.xml
instrument/devices/usaxs_fly_scan.py:sfs = SaveFlyScan("/tmp/sfs.h5", config_file="usaxs_support/saveFlyData.xml")
instrument/devices/usaxs_fly_scan.py:        self.saveFlyData_config = "usaxs_support/saveFlyData.xml"
instrument/devices/usaxs_fly_scan.py:usaxs_flyscan.saveFlyData_config = "/share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml"
usaxs_support/bs_sfd_100.py:saveFlyData_config = "saveFlyData.xml"
usaxs_support/builder_362.py:updates into file: /share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml
usaxs_support/nexus.py:XML_CONFIGURATION_FILE = os.path.join(path, 'saveFlyData.xml')
usaxs_support/saveFlyData.py:XML_CONFIGURATION_FILE = os.path.join(path, 'saveFlyData.xml')
usaxs_support/saveFlyData.py:/APSshare/anaconda/x86_64/bin/python ./saveFlyData.py ./test.h5 ./saveFlyData.xml
usaxs_support/saveFlyData.xml:    /share1/AreaDetectorConfig/FlyScan_config/saveFlyData.xml
usaxs_support/saveFlyData.xsd:  by analysis of example saveFlyData.xml file
usaxs_support/tester_100.py:PVFILE = "saveFlyData.xml"
```

prjemian, 2022-02-01:
Note the functions with the messages reported above are both in `/home/beams/USAXS/.ipython/profile_bluesky/startup`:

```
usaxs_support/saveFlyData.py:    def preliminaryWriteFile(self):
usaxs_support/saveFlyData.py:    def saveFile(self):
```

closed 2022-02-01 by prjemian