APS-USAXS/usaxs-bluesky-ended-2023 #547: BS '<' error

jilavsky, 2022-02-01:
Started in new terminal, BlueSky loads. 
Tried to start data collection. Something about 
_TypeError: '<' not supported between instances of 'KohzuSoftPositionerTuple' and 'float'_
stops data collection... Cannot proceed. Switching to spec. Call me when ready, please. 

```

I Tue-12:06:03 - memory report: pmem(rss=310996992, vms=3479494656, shared=49659904, text=2560000, lib=0, data=2960629760, dirty=0)
I Tue-12:06:04 - Command file verified
I Tue-12:06:04 - Measuring background for: PD_USAXS
                                                                                                                                                                               
I Tue-12:06:05 - PD_USAXS range=4 gain=1e12 bkg=99.0 +/- 196.0
I Tue-12:06:06 - I0_USAXS range=4 gain=1e9 bkg=35.8 +/- 0.9797958971132711
I Tue-12:06:06 - I00_USAXS range=4 gain=1e9 bkg=1.0 +/- 0.0
                                                                                                                                                                               
I Tue-12:06:07 - PD_USAXS range=3 gain=1e10 bkg=1.0 +/- 0.0
I Tue-12:06:07 - I0_USAXS range=3 gain=1e8 bkg=6.6 +/- 11.2
I Tue-12:06:07 - I00_USAXS range=3 gain=1e8 bkg=1.0 +/- 0.0
                                                                                                                                                                               
I Tue-12:06:09 - PD_USAXS range=2 gain=1e8 bkg=1.0 +/- 0.0
I Tue-12:06:09 - I0_USAXS range=2 gain=1e7 bkg=28.2 +/- 0.39999999999999997
I Tue-12:06:09 - I00_USAXS range=2 gain=1e7 bkg=1.0 +/- 0.0
                                                                                                                                                                               
I Tue-12:06:11 - PD_USAXS range=1 gain=1e6 bkg=1.0 +/- 0.0
I Tue-12:06:11 - I0_USAXS range=1 gain=1e6 bkg=7.6 +/- 10.461357464497615
I Tue-12:06:11 - I00_USAXS range=1 gain=1e6 bkg=1.0 +/- 0.0
                                                                                                                                                                               
I Tue-12:06:13 - PD_USAXS range=0 gain=1e4 bkg=1.0 +/- 0.0
I Tue-12:06:13 - I0_USAXS range=0 gain=1e5 bkg=29.0 +/- 0.0
I Tue-12:06:13 - I00_USAXS range=0 gain=1e5 bkg=1.0 +/- 0.0
TypeError: '<' not supported between instances of 'KohzuSoftPositionerTuple' and 'float'

In [6]: 
```

prjemian, 2022-02-01:
Also noticed here: https://github.com/APS-USAXS/ipython-usaxs/issues/516#issuecomment-1026098231

prjemian, 2022-02-01:
final steps of that report:

```python
trd_autorange_controls_ranges_gain0_background -> 1.0
trd_autorange_controls_ranges_gain0_background_error -> 0.0
I Mon-12:48:16 - TR diode range=0 gain=1e5 bkg=1.0 +/- 0.0
scaler0_preset_time -> 0.025
scaler0_auto_count_delay -> 0.02500000037252903
TypeError: '<' not supported between instances of 'KohzuSoftPositionerTuple' and 'float'
```

prjemian, 2022-02-01:
https://github.com/APS-USAXS/ipython-usaxs/blob/8c149024007334bf67f836fcc7c3ae2e87cd8e9b/profile_bluesky/startup/instrument/plans/command_list.py#L390-L400

calls https://github.com/APS-USAXS/ipython-usaxs/blob/8c149024007334bf67f836fcc7c3ae2e87cd8e9b/profile_bluesky/startup/instrument/plans/command_list.py#L403 https://github.com/APS-USAXS/ipython-usaxs/blob/8c149024007334bf67f836fcc7c3ae2e87cd8e9b/profile_bluesky/startup/instrument/plans/command_list.py#L457

calls https://github.com/APS-USAXS/ipython-usaxs/blob/8c149024007334bf67f836fcc7c3ae2e87cd8e9b/profile_bluesky/startup/instrument/plans/command_list.py#L106

look for `scaler0` to be counted

prjemian, 2022-02-01:
The problem is with a comparison of monochromator value.  Looking at: https://github.com/APS-USAXS/ipython-usaxs/blob/8c149024007334bf67f836fcc7c3ae2e87cd8e9b/profile_bluesky/startup/instrument/devices/monochromator.py#L58-L59

need to search for comparisons that involve `.dcm`:

```bash
usaxs@usaxscontrol ~/.../startup/instrument $ git grep ".dcm"
devices/monochromator.py:    dcm = Component(KohzuSeqCtl_Monochromator, "9ida:")
devices/permit.py:in user calc. This would fail for tune_dcmth and other macros, which may take
plans/axis_tuning.py:    if monochromator.dcm.energy.get() < 10.99:  # ~ 10 keV for Si 220 crystals
plans/axis_tuning.py:    elif 10.99 <= monochromator.dcm.energy.get() < 12.99:   # Si 220 crystals
plans/axis_tuning.py:    elif 12.99 <= monochromator.dcm.energy.get() < 18.1:   # Si 220 crystals
plans/axis_tuning.py:    elif 18.1 <= monochromator.dcm.energy.get() < 20.8:   # Si 220 crystals
plans/axis_tuning.py:    elif 20.8 <= monochromator.dcm.energy.get():   # Si 220 crystals
plans/filters.py:    if monochromator.dcm.energy.get() < 12.1:
plans/filters.py:    elif monochromator.dcm.energy.get() < 18.1:
plans/scans.py:    # angle_offset = q2angle(q_offset, monochromator.dcm.wavelength.get())
plans/scans.py:    startAngle = terms.USAXS.ar_val_center.get()- q2angle(terms.USAXS.start_offset.get(),mon
plans/scans.py:    endAngle = terms.USAXS.ar_val_center.get()-q2angle(terms.USAXS.finish.get(),monochromato
plans/scans.py:    # angle_offset = q2angle(q_offset, monochromator.dcm.wavelength.get())
```

prjemian, 2022-02-01:
Looking in [`apstools`](https://github.com/BCDA-APS/apstools/blob/1de05de8603c8a383e3604001d4c59ce2d5b3d61/apstools/devices/kohzu_monochromator.py#L86-L96):

```python
class KohzuSeqCtl_Monochromator(Device):
    """
    synApps Kohzu double-crystal monochromator sequence control program

    .. index:: Ophyd Device; KohzuSeqCtl_Monochromator
    """

    energy = Component(KohzuSoftPositioner, "BraggE", kind="hinted")
    theta = Component(KohzuSoftPositioner, "BraggTheta", kind="normal")
    # lambda is reserved word in Python, can't use it for wavelength
    wavelength = Component(KohzuSoftPositioner, "BraggLambda", kind="normal")
```

and there is the problem.  The `.dcm.energy` is a [KohzuSoftPositioner](https://github.com/BCDA-APS/apstools/blob/1de05de8603c8a383e3604001d4c59ce2d5b3d61/apstools/devices/kohzu_monochromator.py#L25-L29) (now), not a Signal.  Replace `.get()` with `.position`.  This line:

```python
    if monochromator.dcm.energy.get() < 10.99:  # ~ 10 keV for Si 220 crystals
```

should be changed to:

```python
    if monochromator.dcm.energy.position < 10.99:  # ~ 10 keV for Si 220 crystals
```

NOTE:  The change to make Kohzu energy, theta, & wavelength into positioners was in [apstools 1.5.3](https://github.com/BCDA-APS/apstools/blob/main/CHANGES.rst#new-features-andor-enhancements-1).  We did not see that yet since USAXS was running apstools 1.4 last December.  Now upgraded to latest apstools 1.6.1.

### software versions in last run from December 2021

```python
In [4]: cat = databroker.catalog["9idc_usaxs_retired_2022_01_14"]

In [5]: run = cat.v2[-1]

In [6]: run.metadata["start"]["versions"]
Out[6]: 
{'bluesky': '1.6.7',
 'ophyd': '1.6.0',
 'databroker': '1.2.0',
 'apstools': '1.4.1',
 'epics': '3.4.3',
 'h5py': '2.10.0',
 'numpy': '1.19.2',
 'matplotlib': '3.3.2',
 'spec2nexus': '2021.1.8',
 'pyRestTable': '2020.0.3',
 'pymongo': '3.11.2'}
```

prjemian, 2022-02-01:
### software versions today

```python
In [1]: RE.md["versions"]
Out[1]: 
{'bluesky': '1.8.2',
 'ophyd': '1.6.3',
 'databroker': '1.2.5',
 'apstools': '1.6.1',
 'epics': '3.5.0',
 'h5py': '3.6.0',
 'numpy': '1.21.2',
 'matplotlib': '3.5.0',
 'spec2nexus': '2021.1.8',
 'pyRestTable': '2020.0.3',
 'pymongo': '3.12.0'}
```

prjemian, 2022-02-01:
### demonstration

```python
In [2]: monochromator.dcm.energy.position
Out[2]: 21.000047131358706

In [3]: monochromator.dcm.energy.get()
Out[3]: KohzuSoftPositionerTuple(setpoint=21.000047131358706, readback=21.000047131358706, done=0)
```

prjemian, 2022-02-01:
It's an upgrade detail I missed.

commit 21801ba1 (2022-02-01): FIX #547
closed 2022-02-01 by prjemian