APS-USAXS/usaxs-bluesky-ended-2023 #551: Step scanning Nexus file changed?

jilavsky, 2022-02-15:
My Igor code is now failing on step scanned USAXS files from BS. Looks to me that instrument/monochromator disappeared and now we have instrument/source. My code cannot find energy and similar parameters. 
See 02_15_Rabe_usaxs with AirBlank_0006 being flyscan and AirBlank_0008 being step scan. 
Is this correct change and permanent change (and I will fix Igor code) or is this bug and we need to fix this on BS side ?

prjemian, 2022-02-15:
Right.  Not finding NeXus structure `/entry/instrument/monochromator`:

```
/entry/instrument                                                                                                            OPTIONAL NXDL group in data file              not found:  in /entry/instrument/monochromator                                                                             
```

but finding all the right values in the bluesky streams:

```
/entry/instrument/bluesky/streams/baseline/monochromator_dcm_energy                                                          OK       validItemName                        strict pattern: [a-z_][a-z0-9_]*                                                                                           
/entry/instrument/bluesky/streams/baseline/monochromator_dcm_energy                                                          OK       group in base class                  not defined: NXnote/monochromator_dcm_energy                                                                               
/entry/instrument/bluesky/streams/baseline/monochromator_dcm_energy                                                          OK       known NXDL                           NXdata: recognized NXDL specification                                                                                      
/entry/instrument/bluesky/streams/baseline/monochromator_dcm_energy                                                          OK       NeXus base class                     NXdata: known NeXus base class                                                                                             
/entry/instrument/bluesky/streams/baseline/monochromator_dcm_energy                                                          OPTIONAL NXDL field in data file              not found: /entry/instrument/bluesky/streams/baseline/monochromator_dcm_energy/DATA                                        
...
```

Agree, something has changed.  Suspicious of the procedures that create the soft links.

prjemian, 2022-02-15:
This location:

```
/entry/instrument/source                                                                                                     OK       NXDL field in data file              found: /entry/instrument/source/energy                                                                                     
```

This is the content of that group (as seen by *punx*):

```
      source:NXsource
        @NX_class = "NXsource"
        @target = "/entry/instrument/source"
        current --> /entry/instrument/bluesky/streams/baseline/aps_current/value_start
        cycle --> /entry/instrument/bluesky/streams/baseline/aps_aps_cycle/value_start
        energy:NX_INT64[] = 
          @units = "GeV"
        fill_number --> /entry/instrument/bluesky/streams/baseline/aps_fill_number/value_start
        name:NX_CHAR = b'Advanced Photon Source'
          @short_name = "APS"
        probe:NX_CHAR = b'x-ray'
        type:NX_CHAR = b'Synchrotron X-ray Source'
```

The `energy` field is not right.  The others are the expected links to the bluesky stream.

prjemian, 2022-02-15:
Ah, _that_ energy is the storage ring energy.  It's OK.

prjemian, 2022-02-15:
Not seeing a `monochromator` group in `/entry/instrument`:

```python
In [1]: import h5py

In [2]: root = h5py.File("AirBlank_0008.h5", "r")

In [3]: list(root["/entry/instrument"])
Out[3]: ['bluesky', 'slits', 'source']
```

yet the documentation says to expect it (see line 46 here): https://github.com/APS-USAXS/ipython-usaxs/blob/ffa78c33726e6fc470ab1c5ff42f88a51c526281/profile_bluesky/startup/instrument/callbacks/nxwriter_usaxs.py#L38-L47

prjemian, 2022-02-15:
I suspect that the schema for the bluesky streams has changed and this writer was not yet updated or tested.  (Note that `/entry/instrument/bluesky/streams` exists now but line 44 (above) says `/entry/instrument/bluesky_streams`.

The change in the schema allows more data (all streams, not just a select few) from bluesky to be recorded:

```python
In [4]: list(root["/entry/instrument/bluesky"])
Out[4]: ['metadata', 'plan_name', 'streams', 'uid']

In [5]: list(root["/entry/instrument/bluesky/streams"])
Out[5]: ['aps_current_monitor', 'baseline', 'primary']
```

prjemian, 2022-02-15:
Looking at [`apstools.callbacks.nexus_writer.NXWriter.get_stream_link()`](https://github.com/BCDA-APS/apstools/blob/3722eff8cf205ea03e274ac775df963f9ae2b847/apstools/callbacks/nexus_writer.py#L161) to see if we need to supply additional arguments when we call it now...

prjemian, 2022-02-15:
Created callable function from the `get_stream_link()` method to test without the `NXWriter` class itself:

```python
def get_stream_link(root, signal, stream=None, ref=None):
    """
    return the h5 object for ``signal``
    DEFAULTS
    ``stream`` : ``baseline``
    ``key`` : ``value_start``
    """
    stream = stream or "baseline"
    ref = ref or "value_start"
    h5_addr = (
        f"/entry/instrument/bluesky/streams/{stream}/{signal}/{ref}"
    )
    if h5_addr not in root:
        raise KeyError(f"HDF5 address {h5_addr} not found.")
    # return the h5 object, to make a link
    return root[h5_addr]
```

and then call it with one of the lines from `instrument.callbacks.nxwriter_usaxs.py`:

```python
In [9]: get_stream_link(root, "user_data_sample_title")
Out[9]: <HDF5 dataset "value_start": shape (), type "|O">

In [10]: get_stream_link(root, "user_data_sample_title")[()]
Out[10]: b'AirBlank'

In [12]: get_stream_link(root, "user_data_sample_title")[()].decode()
Out[12]: 'AirBlank'

```

prjemian, 2022-02-15:
`uascan` file saving needs to be update to current apstools support.

Where is the monochromator group saved?

prjemian, 2022-02-15:
See here: https://github.com/BCDA-APS/apstools/blob/3722eff8cf205ea03e274ac775df963f9ae2b847/apstools/callbacks/nexus_writer.py#L380

prjemian, 2022-02-15:
Here is the stream value using the test function:

```python
In [18]: get_stream_link(root, "monochromator_dcm_energy")[()]
Out[18]: 21.000047131358706
```

prjemian, 2022-02-15:
In `apstools`, the `monochromator` group expects to find these keys:

```python
keys = "wavelength energy theta y_offset mode".split()
```

but we don't have the `y_offset` key in our `baseline` stream:

```python
In [25]: [k for k in nm if k.find("offset") >= 0]
Out[25]: 
['terms_SAXS_ax_limit_offset',
 'terms_SAXS_dx_limit_offset',
 'terms_SAXS_y_limit_offset',
 'terms_SAXS_z_limit_offset',
 'terms_USAXS_start_offset',
 'terms_WAXS_x_limit_offset']
```

Could add this signal to our baseline or make a local override of `write_monochromator()` that picks different keys.  (Could add `keys` kwarg to the `apstools` support in a future update; that would be more flexible).

Going with the local override for now.

prjemian, 2022-02-15:
The logs are not informative to any possible error information:

```python
|2022-02-15 08:44:42.333|INFO|32635|bluesky-session|nxwriter|15|MainThread| - /home/beams/USAXS/.ipython/profile_bluesky/startup/instrument/callbacks/nxwriter.py
|2022-02-15 08:44:42.334|INFO|32635|bluesky-session|nxwriter_usaxs|16|MainThread| - /home/beams/USAXS/.ipython/profile_bluesky/startup/instrument/callbacks/nxwriter_usaxs.py
```

prjemian, 2022-02-15:
@jilavsky : This code is in place and ready to test, when you have time available.

jilavsky, 2022-02-18:
Sunday afternoon. I'll let you know.

prjemian, 2022-02-18:
Links can be added in the existing files if you wish.  We could try that first on a _copy_ of the file to be modified.

jilavsky, 2022-02-18:
These were purely test scans, no need for the data. I was simply testing, that all _**still works**_. You see the trust level I have by now ;-)

jilavsky, 2022-02-20:
Tested, works, Igor data reduction is happy.

closed 2022-02-20 by jilavsky