APS-USAXS/usaxs-bluesky-ended-2023 #552: wrong folder after newUser()

prjemian, 2022-02-15:
Observation: I started BS in the same terminal again and all seemed to run... EXCEPT, when I tried to collect SAXS/WAXS data, folders and locations which were suppose to be created by newUser were wrong. Instead to 02_15_Rabe/... data were saved on 2022-02_saxs and _waxs. This was AFTER I restarted BS, so something in environment is hanging and corruption future BS sessions. I created new terminal and started BS there and all was OK.

_Originally posted by @jilavsky in https://github.com/APS-USAXS/ipython-usaxs/issues/546#issuecomment-1040362377_

prjemian, 2022-02-20:
@jilavsky Is this still a problem today?  Can you reproduce it?  Or must we assume that the terminal's environment was corrupted (evidence: cleared in new terminal).  If the latter, it's a one-time problem that is difficult to reproduce or repair.

jilavsky, 2022-02-20:
Did not crash BlueSky today, therefore could not test this. 

May be I was not clear - this issue happened after BS crashed - in #546 . After restart of BS in the same terminal session where BS crashed, newUser to failed to set paths correctly. 

Seems unexpected to me, but crash & subsequent start of BS is not really clean restart of BS. To clean the slate we need to create a new terminal session.  Is this expected? I am sure this is important info for our staff, creating new terminal session and navigating to correct work directory is enough hassle that it will not be done unless drilled in as necessary.

prjemian, 2022-02-21:
Too much of this is related to one single observation to create any general advice.

prjemian, 2022-02-21:
Marking this as _won't fix_ (there is no _can't fix_ label) since it is not testable (requires unpredictable crash of Bluesky).

closed 2022-02-21 by prjemian