APS-USAXS/usaxs-bluesky-ended-2023 #557: WAXS image not collected again

jilavsky, 2022-04-12:
We have seen this. WAXS image not collected for some reason. Run fine in spec. Optical camera is working from epics adn save spec images. SAXS image was collected and used, but  

NOTE:
WAXS AD shows red: ERROR CAPTURE NOT SUPPORTED IN SINGLE MODE. 
we changed to single mode when WAXS could not save data in spec when it was in capture mode. When I changed to Capture model BS seems to be saving images. But now spec will not be working... 
SAXS detector is in single mode. Why is WAXS detector in capture mode??? 

```
I Tue-08:23:08 - Pinhole SAXS is in position                                                                                                                              
                                                                                                                                                                          
                                                                                                                                                                          
I Tue-08:23:11 - Creating technique directory: /share1/USAXS_data/2022-04/04_12_Bavand/04_12_Bavand_saxs
I Tue-08:23:11 - Area Detector HDF5 file: /share1/USAXS_data/2022-04/04_12_Bavand/04_12_Bavand_saxs/AgBehenateLaB6_0001.hdf
I Tue-08:23:11 - Pilatus computer Area Detector HDF5 file: /mnt/usaxscontrol/USAXS_data/2022-04/04_12_Bavand/04_12_Bavand_saxs/AgBehenateLaB6_0001.hdf
                                                                                                                                                                          
                                                                                                                                                                          
D Tue-08:23:13 - gain=10000000.0  rate: 22160.0  max: 950000  converged=[True, True]
                                                                                                                                                                          
D Tue-08:23:13 - gain=100000000.0  rate: 224760.0  max: 950000  converged=[False, True]
                                                                                                                                                                          
D Tue-08:23:13 - gain=100000000.0  rate: 224580.0  max: 950000  converged=[True, True]
D Tue-08:23:13 - gain=10000000.0  rate: 453620.0  max: 950000  converged=[True, True, True, True]
                                                                                                                                                                          
I Tue-08:23:19 - Measured SAXS transmission values, pinDiode cts =515716.000000 with gain 100000.0 and I0 cts 181947.0 with gain 10000000.0                               
                                                                                                                                                                          
D Tue-08:23:20 - gain=100000000.0  rate: 652300.0  max: 950000  converged=[True, True]
                                                                                                                                                                          
W Tue-08:23:31 - Could not take sample image:path=/share1/USAXS_data/2022-04/04_12_Bavand/04_12_Bavand_saxs, file=AgBehenateLaB6, order#=1, exc=Attempted to set EpicsSignalWithRBV(read_pv='9idFLY2:JPEG1:FileTemplate_RBV', name='blackfly_optical_jpeg1_file_template', parent='blackfly_optical_jpeg1', value='%', timestamp=1649769801.3935013, auto_monitor=True, string=True, write_pv='9idFLY2:JPEG1:FileTemplate', limits=False, put_complete=False) to value '%s%s_%4.4d.jpg' and timed out after 10 seconds. Current value is '%'.
D Tue-08:23:31 - areaDetectorAcquire(): OrderedDict([('enable', 1), ('auto_increment', 'Yes'), ('array_counter', 0), ('auto_save', 'Yes'), ('num_capture', 0), ('file_template', '%s%s_%4.4d.hdf'), ('file_write_mode', 'Single'), ('blocking_callbacks', 'No'), ('parent.cam.array_callbacks', 1)])

Transient Scan ID: 12     Time: 2022-04-12 08:23:32
Persistent Unique Scan ID: 'a8ff68c9-0d8b-485f-b157-3cab117230dc'

I Tue-08:23:36 - SAXS: 15.0s remaining
I Tue-08:23:41 - SAXS: 10.0s remaining
                                                                                                                                                                          
I Tue-08:23:53 - Finished SAXS/WAXS data collection in 21.491600036621094 seconds.
I Tue-08:23:53 - I0 value: 13050037.0
I Tue-08:23:53 - memory report: pmem(rss=367165440, vms=4996276224, shared=66904064, text=2560000, lib=0, data=4366647296, dirty=0)
I Tue-08:23:53 - file line 26:       waxsExp        25.3      35.7      1      "AgBehenateLaB6"
I Tue-08:23:53 - Found UsaxsSaxsMode = SAXS in beam
I Tue-08:23:53 - Moving to proper WAXS mode
                                                                                                                                                                          
I Tue-08:24:00 - Removed SAXS from beam position
I Tue-08:24:00 - Moving USAXS out of beam
I Tue-08:24:00 - Removed USAXS from beam position
                                                                                                                                                                          
                                                                                                                                                                          
                                                                                                                                                                          
I Tue-08:24:10 - WAXS is in position
D Tue-08:24:10 - guard slits horizontal difference = 0
D Tue-08:24:10 - guard slits vertical difference = 0
D Tue-08:24:10 - USAXS slits horizontal difference = 8.16e-05
D Tue-08:24:10 - USAXS slits vertical difference = 0.000156
I Tue-08:24:10 - Prepared for WAXS mode
I Tue-08:24:10 - Creating technique directory: /share1/USAXS_data/2022-04/04_12_Bavand/04_12_Bavand_waxs
I Tue-08:24:10 - Area Detector HDF5 file: /share1/USAXS_data/2022-04/04_12_Bavand/04_12_Bavand_waxs/AgBehenateLaB6_0001.hdf
                                                                                                                                                                          
D Tue-08:24:12 - gain=100000000.0  rate: 655160.0  max: 950000  converged=[True, True]
                                                                                                                                                                          
D Tue-08:24:12 - gain=100000000.0  rate: 654140.0  max: 950000  converged=[True, True]
                                                                                                                                                                          
D Tue-08:24:13 - gain=100000000.0  rate: 655700.0  max: 950000  converged=[True, True]
                                                                                                                                                                          
D Tue-08:24:13 - gain=100000000.0  rate: 655720.0  max: 950000  converged=[True, True]
D Tue-08:24:13 - gain=100000.0  rate: 184100.0  max: 950000  converged=[True, True, True, True]
                                                                                                            
                                                                                                                                                                          
W Tue-08:24:23 - Could not take sample image:path=/share1/USAXS_data/2022-04/04_12_Bavand/04_12_Bavand_waxs, file=AgBehenateLaB6, order#=1, exc=Attempted to set EpicsSignalWithRBV(read_pv='9idFLY2:JPEG1:FileTemplate_RBV', name='blackfly_optical_jpeg1_file_template', parent='blackfly_optical_jpeg1', value='%', timestamp=1649769853.7528403, auto_monitor=True, string=True, write_pv='9idFLY2:JPEG1:FileTemplate', limits=False, put_complete=False) to value '%s%s_%4.4d.jpg' and timed out after 10 seconds. Current value is '%'.
D Tue-08:24:23 - areaDetectorAcquire(): OrderedDict([('enable', 1), ('auto_increment', 'Yes'), ('array_counter', 0), ('auto_save', 'Yes'), ('num_capture', 0), ('file_template', '%s%s_%4.4d.hdf'), ('file_write_mode', 'Single'), ('blocking_callbacks', 'No'), ('parent.cam.array_callbacks', 1)])
I Tue-08:24:28 - WAXS: 15.0s remaining
[... cut ...] t after 10 seconds. Current value is '/mnt/usaxscontrol/USAXS_data/2022-04/04_12_Bavand/04_12_Bavand_waxs'.
Stopping further processing of this command list.

Out[8]: :24:34 - memory report: pmem(rss=367022080, vms=5004668928, shared=66904064, text=2560000, lib=0, data=4375040000, dirty=0)
('90f6c04a-57f1-4847-a5b7-8cc506f7046d',
 '32f91aad-df69-461b-bc08-ed9528dad856',
 '8603b044-2848-4d3f-9902-cc1c2701c705',
 'c8a15708-b403-4050-b810-db96746f5c8e',
 'a8ff68c9-0d8b-485f-b157-3cab117230dc')

```

jilavsky, 2022-04-12:
Weird. HDF5 plugin changes between capture and single mode while it is collecting images. But works when it is set in epics to capture, even though during exposure BS changes that to single mode. This is confusing. SAXS stays in single mode all the time.

jilavsky, 2022-04-14:
The change we did yesterday in ordering that dictionary made WAXS detector fail today with "This is not allowed in Capture mode" message. I restarted BS to reset dictionary to old order to make sure things work when we change between spec and BS. The code needs to be more robust on what form it finds AD in and not fault when mode is something it does not expect. More development needed here... 
Spec requires both SAXS and WAXS detectors to be in single mode.

jilavsky, 2022-04-19:
Note, restarted BS and now SAXS is not collected

```
E Tue-08:24:42 - Exception TimeoutError during attempt 1 of 1 of command '('saxsExp', ['25.2', '35.7', '1', 'AgBehenateLaB6'], 23, '      saxsExp        25.2      35.7      1      "AgBehenateLaB6"')''
subject: TimeoutError during attempt 1 of 1 of command '('saxsExp', ['25.2', '35.7', '1', 'AgBehenateLaB6'], 23, '      saxsExp        25.2      35.7      1      "AgBehenateLaB6"')''

date: 2022-04-19 08:24:42.421781
command file: /share1/USAXS_data/2022-04/usaxs.mac
line number: 23
command: ('saxsExp', ['25.2', '35.7', '1', 'AgBehenateLaB6'], 23, '      saxsExp        25.2      35.7      1      "AgBehenateLaB6"')
raw command:       saxsExp        25.2      35.7      1      "AgBehenateLaB6"
attempt: 1 of 1
exception: Attempted to set EpicsSignalWithRBV(read_pv='usaxs_pilatus1:HDF1:Capture_RBV', name='saxs_det_hdf1_capture', parent='saxs_det_hdf1', value=0, timestamp=1650374672.2548933, auto_monitor=True, string=False, write_pv='usaxs_pilatus1:HDF1:Capture', limits=False, put_complete=False) to value 1 and timed out after 10 seconds. Current value is 0.
Stopping further processing of this command list.

Out[3]: :24:42 - memory report: pmem(rss=351531008, vms=4854427648, shared=66240512, text=2560000, lib=0, data=4224798720, dirty=0)
('de9f84a9-6e85-490e-b8d7-eecaccc573fb',
 '5b496826-84aa-47c9-be4a-2dab67587972',
 'f2a23ba5-15fa-4d7c-a162-2a6e92cc150b',
 '4f017db4-35a6-42f0-acd5-9403f89a4548')

```

jilavsky, 2022-04-19:
This is same issue as with WAXS detector we had. SAXS detector is set for spec in Single mode and I get now write error: capture not supported in single mode and that fails to collect data. 
This worked until I restarted BS. Switching back to spec, I cannot fight this many wars at the same time...

prjemian, 2022-10-11:
Show me the waxsdet.hdf1.stage_sigs value

jilavsky, 2022-10-11:
Has to wait until we get the detectors up and running. I asked Nino to address move issues, current configuration is sector specific. I need them to clone old 9ID configuration to 20ID one now. Hopefully tomorrow I can start the detectors and then you can get what you need.

jilavsky, 2022-10-13:
WAXS detector is running. When we can start BS, you can do development.