APS-USAXS/usaxs-bluesky-ended-2023 #563: Configure for new USAXS hardware and 20IDB

jilavsky, 2022-10-11:
We need to reconfigure PVs for new Aerotech drivers and 20IDB PSS system. There is little to bet from obsolete EPS system. Lots of smarts need to be disabled.

jilavsky, 2022-10-16:
Edited stages.py:
changed PVs for motors of 
s_stage, d_stage,m_stage,a_stage,saxs_satge, and waxsx.
@prjemian - we do not have currently ms and as satges, they will be completely redone using different hardware. Should be comment these out completely or find some way of making these innactive. Not sure what impact of removing ms and as will be in rest of the code. 
Also, added waxs2x - we now have TWO waxs x stages. the second will be used int eh future for Dexela.  I assume this is for now just placeholder for proper stage and motor name. 
For completeness, here are new PVs for these motors:
DX | 9idcAERO:m1
DY | 9idcAERO:m2
WAXSX | 9idcAERO:m3
AX | 9idcAERO:m4
AY | 9idcAERO:m5
AR | 9idcAERO:m6
WAXS2X | 9idcAERO:m7
SX | 9idcAERO:m8
SY | 9idcAERO:m9
MX | 9idcAERO:m10
MY | 9idcAERO:m11
MR | 9idcAERO:m12
SAXSX | 9idcAERO:m13
SAXSZ | 9idcAERO:m14
SAXSY | 9idcAERO:m15

jilavsky, 2022-10-16:
Changed slits.py - new motor assignment is that all slits motors (blades, centers and gaps) are on 9idcLAX:m58:c1, Guard slits x and y are on 9idcLAX:m58:c0:m7 and m6. This is simpel PV replacement.

jilavsky, 2022-10-16:
@prjemian - need solution for monochromator.py, 
These are new PVs:
feedback is same, 9idcLAX:fbe:omega etc. no change.
there is no info for cryo_level. 
there is no PV for temperature
there is PV for energy: 20id:monoE, this is energy value in eV (= 21keV reads as 21000) and it is NOT motor. I believe energy will change, if we write new value in this field, but have not tried that yet. 

What do we do with this ?

jilavsky, 2022-10-16:
diagnostics.py - changed PVs for 20ID station A and B
removed station C
removed c_station_enabled, this is enabled always. 
removed BLEPS & FEEPS, this one does not have PV to tell us anything (this is obsolete device)  
note, this removed from device DiagnosticsParameters BL_EPS and FE_EPS.

jilavsky, 2022-10-16:
permit.py changed operations_in_9idc to return always true.

jilavsky, 2022-10-16:
shutters.py
changed PVs for 20id: FE shutter autoopen is 9idcLAX:rShtrA, 
for FE and mono shutters, we have different system - there are PVs: 20id:shutter0_opn and 20id:shutter0_cls which open (write 1 in opn) and close (write 1 in cls). Same for mono shutter which is shutter1: 20id:shutter1_opn, and 20id:shutter1_cls. @prjemian I have no idea what to do for FE_shutter and mono_shutter in here. Looks like there is only 1 PV used to open/close here.

jilavsky, 2022-10-16:
**white_beam_ready_calc**.py - changed beamline PVs
**protection_plc**.py is not used, this function is done on Automation1 controller directly. Stages will refuse to move unless safe. This is under development. We could hard wire same logic here, but seems excessive. I commented out installing suspender due to PLC. 
**miscallaneous**.py - removed camy, fuel_spray_bit, tcam, tension.   Not used at this time.

jilavsky, 2022-10-16:
**general_terms**.py updated PVs to new logic
DY0 -> 9idcLAX:Diode_dx
these changed from *Y0 to *X0 - we will scan in X direction. 
AX0 = Component(EpicsSignal,                      "9idcLAX:SAXS:ax_in")
DX0 = Component(EpicsSignal,                      "9idcLAX:SAXS:dx_in")
@prjemian USAXS.AX0 and USAXS.DX0 now share PV with SAXS.ax_in and SAXS.dx_in, is that problem? 

I **think** I fixed this, but this is the most likely place of any bugs.

jilavsky, 2022-10-16:
aps_source.py - changed to 20ID for undulator. 
suspenders.py - I think this does not need changing?
trajectories.py - we will scan ar, ax, and dy, just changed the names to 9idcLAX:traj*:M1Traj to designate that we scan ax and dx.

jilavsky, 2022-10-16:
**usaxs_fly_scan**.py - 
changed ay and dy to ax and dx (lines 87/88, lines 175/176, 248/249) Looks like all what was needed.

jilavsky, 2022-10-16:
Plans changes
**move_instrument.py** - changed moving targets for ay and dy into terms.SAXS.ay_in and dy_in. removed AY0 and DY0. Dee comment on general terms  where I added AX0 and DX0, which are same 

**sample_transmission.py** - here we need to move horizontally to get diode in the beam (USAXS/SAXS) so changed code as needed...

**scans.py** - changed step scan to move horizontally. Line 324/325 changed to DX0 and AX0. More similar blocks. 

**uascan.py** - changed to move in x direction.

jilavsky, 2022-10-16:
**reset_USAXS.py** - changed to move in x direction

jilavsky, 2022-10-19:
TODO: BS cannot open B shutter. Something is wrong. 
After abort from "unable to open shutter in 5 seconds" RE.abort() does not clean enough to enable operations. Needs restart.

prjemian, 2022-10-19:
@jilavsky emailed:

> BS HDF5 data saving will need to be corrected. It is missing monochromator_dcm_theta and throws error. Is that using /share1/AreaDetectorConfig/Flyscan_config/saveFlyData.xml? I fixed there the dcm_theta is looking at 20id:m41.RBV
> 
> I suspect BS step scan is using something else, it is reading data from DataBroker...

jilavsky, 2022-10-19:
TODO: AR stage seems to sometimes need more time to reach position. DUmbed down in epics for now and it seems to work, but we may need to add more retries for this stege?

jilavsky, 2022-10-19:
AR stage still having issues. Mid way scanning I get following error:

```
generator uascan ['af763427'] (scan num: 68)                                                                                                
D Wed-14:44:58 - write_entry of file: /share1/USAXS_data/2022-10/10_19_Test/10_19_Test_usaxs/Blank_0199.h5                                  
E Wed-14:44:58 - Exception FailedStatus during attempt 1 of 1 of command '('USAXSscan', ['20', '20', '0', 'Blank'], 22, '\t  USAXSscan       20      20      0      "Blank"')''
subject: FailedStatus during attempt 1 of 1 of command '('USAXSscan', ['20', '20', '0', 'Blank'], 22, '\t  USAXSscan       20      20      0      "Blank"')''

date: 2022-10-19 14:44:58.053453
command file: /share1/USAXS_data/2022-10/usaxs.mac
line number: 22
command: ('USAXSscan', ['20', '20', '0', 'Blank'], 22, '\t  USAXSscan       20      20      0      "Blank"')
raw command: 	  USAXSscan       20      20      0      "Blank"
attempt: 1 of 1
exception: MoveStatus(done=True, pos=a_stage_r, elapsed=0.3, success=False, settle_time=0.0)
Stopping further processing of this command list.
```
to the best of my abilities this looks like ar stage is not reaching its position in time allocated. I can try to retune the stage, but it would be nice if this specific stage gave it bit more time. I have relaxed the conditions in spec - set number of retries to 10 (was lower to speed up spec scanning) and increased Retry deadband. We will see if it helps.

jilavsky, 2022-10-19:
May not be problem. I retuned the stage and seems to be working now.

jilavsky, 2022-10-20:
Still issue, had to dumb down motor to 2e-5 deg Retry deadband and 10 retires. routinely have 3 retries, sometimes more.

prjemian, 2022-10-20:
Servo motor?

jilavsky, 2022-10-20:
PID loop driven motors (all of them) against encoder. Encoder resolution is officially 3e-7 degrees for rotational stages, official minimum step size is 0.03 arc seconds (8e-6 degrees). Sometimes may take longer time to settle at position. This is expected (and not problem when flyscanning)... Tuning helped, but seems like larger steps cause stage to overshoot. I might be able to improve optimization more.

prjemian, 2022-10-25:
We're not done with this issue, so re-open it.

PR #565 (2022-10-25): configure for new usaxs hardware and 20 idb
- close #563