APS-USAXS/usaxs-bluesky-ended-2023 #568: Fix scalers with SAXS/WAXS

jilavsky, 2022-10-30:
We have unexpected behavior of our S/WAXS code. 
We expect that scaler0 and scaler1 collect I0 (ungated and gated) during AD exposure. Our code sets time in lines 849-865 parameters needed for this collection. Lines 867/869 triggers the data collection on scaler0 and scaler1. Lines 865-899 stop counting and collect I0 values for AD to save in Nexus writer. 

Unluckily, this fails: Currently we set the values and trigger the scalers counting, but BS seems to stop scaler0 and scaler1 counting during SAXS/SAXS AD data collection. So we hav eshort counting before AD triggers and short collection after it triggers, but we completely miss the time when we need these scalers to be collecting data. 

I suspect that BS (bps.count) stops the scaler0 and scaler1 during the counting since it is not instructed to collect the data on them and does not ignore them as  spec used to so (and BS used to do in prior incarnation) .

prjemian, 2022-10-30:
> Assigned #568 <https://github.com/APS-USAXS/usaxs-bluesky/issues/568> to
> @prjemian <https://github.com/prjemian>.

Bluesky will not take control of any detector or other devices but
described in the plan.  With have to investigate the code to determine what
is happening. I do not suspect bluesky is telling scalers not to count if
they are not given as detectors in the plan.

jilavsky, 2022-10-30:
Something is fishy here, this is how both scalers end after 20 seconds WAXS exposure.
Alos, we still have some issue with time, 20 seconds WAXS exposure takes 21.7 seconds. Even though this is 10th WAXS exposure in a row - with same times. 
[image]

jilavsky, 2022-10-30:
More observations. We get about 0.5 seconds collection before 20 second exposure then Scalers go to Done, then after exposure we get another 1-1.4 seconds separate count. While AD is Collecting, scalers are in Done state, something had to push Done button on them. 

[image]

[image]

jilavsky, 2022-10-30:
We are looking for two things:
1. Someone stops the Scalers after we start it before exposing AD
2. Someone again starts scalers after exposing AD....

Looking at SAXS (WAXS is same), this is in scans.py. None of these lines outside areaDetectorAcquire should have any impact on scalers... 

```
    _md['plan_name'] = "SAXS"
    _md["hdf5_file"] = SAXS_file_name
    _md["hdf5_path"] = SAXSscan_path

    yield from record_sample_image_on_demand("saxs", scan_title_clean, _md)

    yield from areaDetectorAcquire(saxs_det, create_directory=-5, md=_md)
    ts = str(datetime.datetime.now())
    yield from bps.remove_suspender(suspend_BeamInHutch)

    saxs_det.hdf1.stage_sigs = old_det_stage_sigs    # TODO: needed? not even useful?
```
So we need to look inside area_detector.py in areaDetectorAcquire, but none of these look suspicious. The only actionable lines seem to be 68-70:

```
    bec.disable_table()
    yield from bp.count([det], md=_md)          # TODO: SPEC showed users incremental progress (1 Hz updates) #175
    bec.enable_table()
```

prjemian, 2022-10-31:
These lines start the two scalers counting using ophyd methods (unknown to the RE).  The `stage()`/`unstage()` process is not used. https://github.com/APS-USAXS/usaxs-bluesky/blob/a803031b8c806c74949ff53fab1ef4cc3cb2f678/instrument/plans/scans.py#L867-L871

prjemian, 2022-10-31:
The text `scaler` does not occur in the file where `record_sample_image_on_demand()` is defined nor does it occur in the file where`areaDetectorAcquire()` is defined.

This shoots down the theory that the `RE` might be staging the scalers.  However, this can be checked in my simulators here.

prjemian, 2022-10-31:
See a suggestion: https://github.com/APS-USAXS/usaxs-bluesky/issues/569#issuecomment-1296419097
and reply that follows.

prjemian, 2022-10-31:
In my simulator, using scaler `scaler` and area detector `adsimdet` (with HDF plugin disabled), 

<details>
<summary>test code</summary>

```python
from bluesky import plan_stubs as bps
from bluesky import plans as bp
from bluesky import preprocessors as bpp
from instrument.devices import adsimdet
from instrument.devices import scaler1

for k in "acquire_time acquire_period".split():
    if k in adsimdet.cam.stage_sigs:
        adsimdet.cam.stage_sigs.pop(k)

def usaxs_issue_568():
    @bpp.monitor_during_decorator([scaler1.count])
    def _acquire_image():
        yield from bp.count([adsimdet])

    yield from bps.mv(
        scaler1.preset_time, 30,
        adsimdet.cam.acquire_time, 5,
        adsimdet.cam.acquire_period, 5.01,
    )
    yield from bps.mv(scaler1.count, 1)
    yield from _acquire_image()
    yield from bps.mv(scaler1.count, 0)
```

</details>

this the test run:

```python
In [6]: %run -i /tmp/usaxs_568.py

In [7]: RE(usaxs_issue_568())

Transient Scan ID: 366     Time: 2022-10-31 11:52:28
Persistent Unique Scan ID: '1464876a-7932-4975-a407-b4659169e710'
New stream: 'scaler1_count_monitor'
New stream: 'primary'
+-----------+------------+
|   seq_num |       time |
+-----------+------------+
|         1 | 11:52:33.2 |
+-----------+------------+
generator count ['1464876a'] (scan num: 366)
Out[7]: ('1464876a-7932-4975-a407-b4659169e710',)

In [8]: RE(usaxs_issue_568())

Transient Scan ID: 367     Time: 2022-10-31 11:52:40
Persistent Unique Scan ID: 'fbf66649-0f40-4563-8cf4-606c1bfb252c'
New stream: 'scaler1_count_monitor'
New stream: 'primary'
+-----------+------------+
|   seq_num |       time |
+-----------+------------+
|         1 | 11:52:45.5 |
+-----------+------------+
generator count ['fbf66649'] (scan num: 367)
Out[8]: ('fbf66649-0f40-4563-8cf4-606c1bfb252c',)
```

Now, examine the monitor stream to see if the scaler was stopped or restarted during area detector acquisition:

```python
In [9]: run = cat[-1]

In [10]: run
Out[10]: 
BlueskyRun
  uid='fbf66649-0f40-4563-8cf4-606c1bfb252c'
  exit_status='success'
  2022-10-31 11:52:40.489 -- 2022-10-31 11:52:45.520
  Streams:
    * scaler1_count_monitor
    * primary

In [11]: run.scaler1_count_monitor.read()
Out[11]: 
<xarray.Dataset>
Dimensions:        (time: 1)
Coordinates:
  * time           (time) float64 1.667e+09
Data variables:
    scaler1_count  (time) int64 1
```

Only one event in the scaler's monitor stream, that's the initial value.

prjemian, 2022-10-31:
note: If we add the scaler to the `bp.count()` call, then bluesky will wait until both adsimdet and scaler1 finish. not a good model of the original problem.

prjemian, 2022-10-31:
So, our data acquisition procedure should be similar to this prototype I used for testing:

```python
def usaxs_issue_568():
    @bpp.monitor_during_decorator([scaler1.count])
    def _acquire_image():
        yield from bp.count([adsimdet])

    yield from bps.mv(
        scaler1.preset_time, 30,
        adsimdet.cam.acquire_time, 5,
        adsimdet.cam.acquire_period, 5.01,
    )
    yield from bps.mv(scaler1.count, 1)
    logger.info("scaler1.time: %s", scaler1.time.get())

    yield from _acquire_image()
    logger.info("scaler1.time: %s", scaler1.time.get())

    yield from bps.mv(scaler1.count, 0)
    logger.info("scaler1.time: %s", scaler1.time.get())
```

prjemian, 2022-10-31:
Code changes for this issue should be done in a branch, to allow immediate rollback (switching branches) until the code is accepted and merged.

prjemian, 2022-11-01:
**Aha!**  The shutters are controlled by a user transform in EPICS.

prjemian, 2022-11-01:
[image]

prjemian, 2022-11-01:
note: this was a copy of the settings from `9idcLAX:userTran1`, the transform that is used.

prjemian, 2022-11-01:
The values in the transform should be changed from 1 to 0, changing the logic of how the scalers are triggered (and fixing this issue).  Because of this transform, it is not necessary for bluesky to trigger either scaler.

We plan to drop the branch and make this last change in `master` branch.

prjemian, 2022-11-01:
The fix works as tested!

closed 2022-11-01 by prjemian