APS-USAXS/usaxs-bluesky-ended-2023 #569: Are these lines correct?

jilavsky, 2022-10-31:


jilavsky, 2022-10-31:
Searching through the code, I think we need to reproduce lines 130-132 after line 141 in this code:

https://github.com/APS-USAXS/usaxs-bluesky/blob/a803031b8c806c74949ff53fab1ef4cc3cb2f678/instrument/plans/sample_transmission.py#L105:~:text=md%5B%22plan_name%22%5D%20%3D%20%22measure_SAXS_Transmission,ti_filter_shutter%2C%20%22close%22%2C

prjemian, 2022-10-31:
https://github.com/APS-USAXS/usaxs-bluesky/blob/a803031b8c806c74949ff53fab1ef4cc3cb2f678/instrument/plans/sample_transmission.py#L130-L142

prjemian, 2022-10-31:
Comments belong to #568

prjemian, 2022-10-31:
not part of AD image acquisition

prjemian, 2022-10-31:
Why should these lines be reproduced after line 141?

Here is the last line any of these are used: https://github.com/APS-USAXS/usaxs-bluesky/blob/a803031b8c806c74949ff53fab1ef4cc3cb2f678/instrument/plans/sample_transmission.py#L134

Lines 130-132 just create local references so that line 134 is not even longer.

jilavsky, 2022-10-31:
Oh, I thought lines 130-132 are storing values for writing transmission stuff into Nexus file. In that case I understand.

closed 2022-10-31 by jilavsky