APS-USAXS/usaxs-bluesky-ended-2023 #570: RE.abort() needs to remove suspenders

jilavsky, 2022-10-31:
RE.abort() does not seem to remove suspenders and operations which do not require beam are still impossible after RE.abort() without beam. RE.abort() should remove the suspenders.

prjemian, 2022-10-31:
https://github.com/APS-USAXS/usaxs-bluesky/blob/a803031b8c806c74949ff53fab1ef4cc3cb2f678/instrument/devices/suspenders.py#L68-L74

https://github.com/APS-USAXS/usaxs-bluesky/blob/a803031b8c806c74949ff53fab1ef4cc3cb2f678/instrument/plans/scans.py#L141

and others

https://github.com/APS-USAXS/usaxs-bluesky/search?q=install_suspender

prjemian, 2022-10-31:
Some suspenders are installed from routine scanning plans.  But there is one that is only installed during initialization.  Restore it with:

```python
from instrument.framework.initialization import suspender_white_beam_ready
RE.install_suspender(suspender_white_beam_ready)
```

prjemian, 2022-10-31:
This issue exists, mostly, because we have added suspenders to a plan and then aborted that plan without the suspenders being removed.  Instead of calling https://github.com/APS-USAXS/usaxs-bluesky/blob/a803031b8c806c74949ff53fab1ef4cc3cb2f678/instrument/plans/scans.py#L141

could we use a suspender decorator that will catch abort and remove the suspender(s) automagically?

prjemian, 2022-10-31:
In [`bluesky.preprocessors` (a.k.a. `bpp`)](https://github.com/bluesky/bluesky/blob/b7d666e65cf4ef556fb46b744c33264c8e3f7507/bluesky/preprocessors.py#L386-L402), there is:

```python
def suspend_wrapper(plan, suspenders):
    """
    Install suspenders to the RunEngine, and remove them at the end.

    Parameters
    ----------
    plan : iterable or iterator
        a generator, list, or similar containing `Msg` objects
    suspenders : suspender or list of suspenders
        Suspenders to use for the duration of the wrapper

    Yields
    ------
    msg : Msg
        messages from plan, with 'install_suspender' and 'remove_suspender'
        messages inserted and appended
    """
```

prjemian, 2022-10-31:
prototype code:

```python
from bluesky import preprocessors as bpp

# ...

    @bpp.suspend_wrapper()
    def steps_that_might_be_suspended():
        yield from this()
        yield from that()

    result = yield from steps_that_might_be_suspended()
    # TODO: might need to return that `result`
```

prjemian, 2022-10-31:
What to do about https://github.com/APS-USAXS/usaxs-bluesky/blob/a803031b8c806c74949ff53fab1ef4cc3cb2f678/instrument/devices/suspenders.py#L74

I think you still want this at the global level?  The only time you would want to remove it is like now, during a shut-down day.

jilavsky, 2022-10-31:
I think the white beam ready suspender should be removed. It was there for historical reasons when we had routine searching for feedback after beam dump. That needed to know the white beam is ready even when it had no I000 value. Our current suspend_BeamInHutch already looks at front end shutter, so it is triggered by FES closing. I suggest removing this altogether and let's see, if we ever need it again. Let's simplify.

commit 0c25f48f (2022-11-01): Closes #570
commit 6fe979be (2022-11-01): FIX #570
closed 2022-11-01 by jilavsky