APS-USAXS/usaxs-bluesky-ended-2023 #574: E-mail sending (?)

jilavsky, 2022-11-01:
This showed few times, I think this is e-mails sending info about beam dump or something else. 

```
Exception in thread Thread-1020 (send):
Traceback (most recent call last):
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/threading.py", line 1009, in _bootstrap_inner
    self.run()
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/threading.py", line 946, in run
    self._target(*self._args, **self._kwargs)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/site-packages/apstools/utils/email.py", line 67, in send
    s = smtplib.SMTP(self.smtp_host)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/smtplib.py", line 255, in __init__
    (code, msg) = self.connect(host, port)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/smtplib.py", line 341, in connect
    self.sock = self._get_socket(host, port, self.timeout)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/smtplib.py", line 312, in _get_socket
    return socket.create_connection((host, port), timeout,
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/socket.py", line 845, in create_connection
    raise err
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/socket.py", line 833, in create_connection
    sock.connect(sa)
TimeoutError: [Errno 110] Connection timed out
```

prjemian, 2022-11-01:
Have any emails be sent with success?  Is it possible that new computer or location has created an unanticipated obstacle>

jilavsky, 2022-11-01:
I have not received any. I assume this was the suspender trying to send "recover from beam dump" message, but I forgot which ones we are sending. This is low priority for now. I wanted you to have documentation on this, looks like lower level BS code than USAXS.

jilavsky, 2022-11-25:
This e-mail sending error is becoming annoying. Seems to be triggered by failed tunes and other actions.

For now I hav e DOISABLED this. this is in devices/emails.py - changed the NOTIFY_ON_... 

NOTE: this is failing in apstools.utils EmailNotifications. Needs fix upstream or something changed...   

```
Exception in thread Thread-10971 (send):
Traceback (most recent call last):
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/threading.py", line 1009, in _bootstrap_inner
    self.run()
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/threading.py", line 946, in run
    self._target(*self._args, **self._kwargs)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/site-packages/apstools/utils/email.py", line 67, in send
    s = smtplib.SMTP(self.smtp_host)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/smtplib.py", line 255, in __init__
    (code, msg) = self.connect(host, port)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/smtplib.py", line 341, in connect
    self.sock = self._get_socket(host, port, self.timeout)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/smtplib.py", line 312, in _get_socket
    return socket.create_connection((host, port), timeout,
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/socket.py", line 845, in create_connection
    raise err
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/socket.py", line 833, in create_connection
    sock.connect(sa)
TimeoutError: [Errno 110] Connection timed out
Exception in thread Thread-11273 (send):
Traceback (most recent call last):
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/threading.py", line 1009, in _bootstrap_inner
    self.run()
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/threading.py", line 946, in run
    self._target(*self._args, **self._kwargs)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/site-packages/apstools/utils/email.py", line 67, in send
    s = smtplib.SMTP(self.smtp_host)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/smtplib.py", line 255, in __init__
    (code, msg) = self.connect(host, port)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/smtplib.py", line 341, in connect
    self.sock = self._get_socket(host, port, self.timeout)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/smtplib.py", line 312, in _get_socket
    return socket.create_connection((host, port), timeout,
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/socket.py", line 845, in create_connection
    raise err
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/socket.py", line 833, in create_connection
    sock.connect(sa)
TimeoutError: [Errno 110] Connection timed out
Exception in thread Thread-11985 (send):
Traceback (most recent call last):
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/threading.py", line 1009, in _bootstrap_inner
    self.run()
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/threading.py", line 946, in run
    self._target(*self._args, **self._kwargs)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/site-packages/apstools/utils/email.py", line 67, in send
    s = smtplib.SMTP(self.smtp_host)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/smtplib.py", line 255, in __init__
    (code, msg) = self.connect(host, port)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/smtplib.py", line 341, in connect
    self.sock = self._get_socket(host, port, self.timeout)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/smtplib.py", line 312, in _get_socket
    return socket.create_connection((host, port), timeout,
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/socket.py", line 845, in create_connection
    raise err
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/socket.py", line 833, in create_connection
    sock.connect(sa)
TimeoutError: [Errno 110] Connection timed out
Exception in thread Thread-12281 (send):
Traceback (most recent call last):
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/threading.py", line 1009, in _bootstrap_inner
    self.run()
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/threading.py", line 946, in run
    self._target(*self._args, **self._kwargs)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/site-packages/apstools/utils/email.py", line 67, in send
    s = smtplib.SMTP(self.smtp_host)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/smtplib.py", line 255, in __init__
    (code, msg) = self.connect(host, port)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/smtplib.py", line 341, in connect
    self.sock = self._get_socket(host, port, self.timeout)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/smtplib.py", line 312, in _get_socket
    return socket.create_connection((host, port), timeout,
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/socket.py", line 845, in create_connection
    raise err
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/socket.py", line 833, in create_connection
    sock.connect(sa)
TimeoutError: [Errno 110] Connection timed out
In [14]: 
```