APS-USAXS/usaxs-bluesky-ended-2023 #582: Non critical error messages

jilavsky, 2023-01-27:
I started to test, moving among geometries. This seems non critical error:

```
In [9]: RE(mode_OpenBeamPath())
I Fri-14:07:22 - Found UsaxsSaxsMode = USAXS in beam
I Fri-14:07:22 - Opening the beam path, moving all components out
I Fri-14:07:22 - Moving SAXS out of beam
I Fri-14:07:22 - Removed SAXS from beam position
I Fri-14:07:22 - Moving WAXS out of beam
I Fri-14:07:22 - Removed WAXS from beam position
I Fri-14:07:22 - Moving USAXS out of beam
                                                                                                                                            
Out[9]: ()                                                                                                                                  
I Fri-14:07:25 - Removed USAXS from beam position
In [10]: RE(mode_USAXS())
W Fri-14:09:40 - !!!!!!!!!!!!!!!
--- Logging error ---
Traceback (most recent call last):
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/logging/__init__.py", line 1100, in emit
    msg = self.format(record)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/logging/__init__.py", line 943, in format
    return fmt.format(record)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/logging/__init__.py", line 678, in format
    record.message = record.getMessage()
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/logging/__init__.py", line 368, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/threading.py", line 966, in _bootstrap
    self._bootstrap_inner()
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/threading.py", line 1009, in _bootstrap_inner
    self.run()
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/threading.py", line 946, in run
    self._target(*self._args, **self._kwargs)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/asyncio/base_events.py", line 600, in run_forever
    self._run_once()
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/asyncio/base_events.py", line 1896, in _run_once
    handle._run()
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/site-packages/bluesky/run_engine.py", line 1523, in _run
    msg = self._plan_stack[-1].send(resp)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/site-packages/bluesky/preprocessors.py", line 1307, in __call__
    return (yield from plan)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/site-packages/bluesky/preprocessors.py", line 1162, in baseline_wrapper
    return (yield from plan_mutator(plan, insert_baseline))
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/site-packages/bluesky/preprocessors.py", line 123, in plan_mutator
    msg = plan_stack[-1].send(ret)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/site-packages/bluesky/preprocessors.py", line 803, in monitor_during_wrapper
    return (yield from plan2)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/site-packages/bluesky/preprocessors.py", line 123, in plan_mutator
    msg = plan_stack[-1].send(ret)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/site-packages/bluesky/preprocessors.py", line 123, in plan_mutator
    msg = plan_stack[-1].send(ret)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/site-packages/bluesky/preprocessors.py", line 861, in fly_during_wrapper
    return (yield from plan2)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/site-packages/bluesky/preprocessors.py", line 123, in plan_mutator
    msg = plan_stack[-1].send(ret)
  File "/home/beams11/USAXS/micromamba/envs/bluesky_2022_3/lib/python3.10/site-packages/bluesky/preprocessors.py", line 123, in plan_mutator
    msg = plan_stack[-1].send(ret)
  File "/home/beams/USAXS/bluesky/instrument/plans/mode_changes.py", line 127, in mode_USAXS
    yield from DCMfeedbackON()
  File "/home/beams/USAXS/bluesky/instrument/plans/mono_feedback.py", line 28, in DCMfeedbackON
    monochromator.feedback.check_position()
  File "/home/beams/USAXS/bluesky/instrument/devices/monochromator.py", line 84, in check_position
    logger.warning(subject, message)
Message: 'USAXS Feedback problem'
Arguments: ('Feedback is very close to its limits.',)
W Fri-14:09:40 - !!!!!!!!!!!!!!!
I Fri-14:09:40 - Found UsaxsSaxsMode = out of beam
I Fri-14:09:40 - Moving to proper USAXS mode
I Fri-14:09:40 - Moving WAXS out of beam
I Fri-14:09:40 - Removed WAXS from beam position
I Fri-14:09:40 - Moving SAXS out of beam
I Fri-14:09:40 - Removed SAXS from beam position
I Fri-14:09:40 - Moving to USAXS mode
                                                                                                                                            
                                                                                                                                            
                                                                                                                                            
Out[10]: ()                                                                                                                                 
                                                                                                                                            
In [11]: RE(mode_OpenBeamPath())
I Fri-14:10:23 - Found UsaxsSaxsMode = USAXS in beam
I Fri-14:10:23 - Opening the beam path, moving all components out
I Fri-14:10:23 - Moving SAXS out of beam
I Fri-14:10:23 - Removed SAXS from beam position
I Fri-14:10:23 - Moving WAXS out of beam
I Fri-14:10:23 - Removed WAXS from beam position
I Fri-14:10:23 - Moving USAXS out of beam
                                                                                                                                            
Out[11]: ()                                                                                                                                 
I Fri-14:10:27 - Removed USAXS from beam position

```

prjemian, 2023-01-27:
> Assigned #582 <https://github.com/APS-USAXS/usaxs-bluesky/issues/582> to
> @prjemian <https://github.com/prjemian>.

> Feedback close it is limits.

This message needs more detail.
Can you confirm with the hardware?

jilavsky, 2023-01-28:
The message is correct, feedback is close to limit as VME was reset.  This is expected message in this case. The point is, that it was suppose to send me e-mail or report error and not throw python exception.

prjemian, 2023-01-28:
Ok, changing the title of this issue.