APS-USAXS/usaxs-bluesky-ended-2023 #583: Tuning fails too often

jilavsky, 2023-02-01:
Tuning error, when the LiveData tool does not get x axis and y axis data, axis is scanned, no result reported. First dx tune failed, dy tone worked. Solution seems ot be restart BS, which is annoying for users. 

```
 RE(tune_diode())
                                                                                                                                                                       
I Wed-14:10:26 - tuning axis: d_stage_x
                                                                                                                                                                       
I Wed-14:10:26 - Tuning axis d_stage_x, current position is 98.97974999999994

Transient Scan ID: 40     Time: 2023-02-01 14:10:26
Persistent Unique Scan ID: '89322aeb-3f0c-46ae-a13d-2932f449d4ac'
New stream: 'baseline'
New stream: 'aps_current_monitor'
New stream: 'primary'                                                                                                                                                  
+-----------+------------+------------+
|   seq_num |       time |   PD_USAXS |
+-----------+------------+------------+
|         1 | 14:10:27.6 |          8 |
|         2 | 14:10:27.9 |         18 |
|         3 | 14:10:28.2 |         36 |
|         4 | 14:10:28.6 |         41 |
|         5 | 14:10:28.9 |         37 |
|         6 | 14:10:29.2 |         23 |
|         7 | 14:10:29.5 |        268 |
|         8 | 14:10:29.9 |       4394 |
|         9 | 14:10:30.2 |      11626 |
|        10 | 14:10:30.5 |      21581 |
|        11 | 14:10:30.8 |      31020 |
|        12 | 14:10:31.1 |      38150 |
|        13 | 14:10:31.5 |      42482 |
|        14 | 14:10:31.8 |      43885 |
|        15 | 14:10:32.1 |      44578 |
|        16 | 14:10:32.4 |      44921 |
|        17 | 14:10:32.8 |      45595 |
|        18 | 14:10:33.1 |      45742 |
|        19 | 14:10:33.4 |      45627 |
|        20 | 14:10:33.7 |      45413 |
|        21 | 14:10:34.0 |      44817 |
|        22 | 14:10:34.4 |      43936 |
|        23 | 14:10:34.7 |      42793 |
|        24 | 14:10:35.0 |      39467 |
|        25 | 14:10:35.3 |      32869 |
|        26 | 14:10:35.6 |      21883 |
|        27 | 14:10:36.0 |      10934 |
|        28 | 14:10:36.3 |       3322 |
|        29 | 14:10:36.6 |         28 |
|        30 | 14:10:36.9 |         19 |
|        31 | 14:10:37.3 |         20 |
|        32 | 14:10:37.6 |         38 |
|        33 | 14:10:37.9 |         47 |
|        34 | 14:10:38.2 |         41 |
|        35 | 14:10:38.6 |         18 |
I Wed-14:10:38 - PeakStats : no max reported
+-----------+------------+------------+
generator tune_dx ['89322aeb'] (scan num: 40)
LivePlot did not get any data that corresponds to the x axis. d_stage_x
LivePlot did not get any data that corresponds to the y axis. PD_USAXS
                                                                                                                                                                       
I Wed-14:10:39 - PeakStats
================ =================
key              result           
================ =================
x                d_stage_x        
y                PD_USAXS         
cen              None             
com              None             
fwhm             None             
min              None             
max              None             
crossings        None             
tune_ok          False            
center           None             
initial_position 98.97974999999994
final_position   98.97974999999994
================ =================

                                                                                                                                                                       
I Wed-14:10:39 - PeakStats : no max reported
I Wed-14:10:39 - axis: d_stage_x
I Wed-14:10:39 - starting position: 98.97973799999991
I Wed-14:10:39 - peak detected: False
                                                                                                                                                                       
I Wed-14:10:39 - tuning axis: d_stage_y
                                                                                                                                                                       
I Wed-14:10:40 - Tuning axis d_stage_y, current position is -15.843211000000224

Transient Scan ID: 41     Time: 2023-02-01 14:10:40
Persistent Unique Scan ID: '76c24fef-dfec-44bd-ac04-128624753022'
New stream: 'baseline'
New stream: 'aps_current_monitor'
New stream: 'primary'                                                                                                                                                  
+-----------+------------+------------+------------+
|   seq_num |       time |  d_stage_y |   PD_USAXS |
+-----------+------------+------------+------------+
|         1 | 14:10:40.9 | -18.843305 |       9138 |
|         2 | 14:10:41.3 | -18.666745 |      14896 |
|         3 | 14:10:41.6 | -18.490272 |      20063 |
|         4 | 14:10:41.9 | -18.313798 |      25080 |
|         5 | 14:10:42.3 | -18.137321 |      30978 |
|         6 | 14:10:42.6 | -17.960852 |      36204 |
|         7 | 14:10:43.0 | -17.784378 |      40017 |
|         8 | 14:10:43.3 | -17.607911 |      41926 |
|         9 | 14:10:43.7 | -17.431445 |      42668 |
|        10 | 14:10:44.0 | -17.254976 |      43056 |
|        11 | 14:10:44.3 | -17.078507 |      43635 |
|        12 | 14:10:44.7 | -16.902045 |      43379 |
|        13 | 14:10:45.0 | -16.725563 |      44210 |
|        14 | 14:10:45.3 | -16.549090 |      44703 |
|        15 | 14:10:45.6 | -16.372624 |      45347 |
|        16 | 14:10:46.0 | -16.196144 |      45831 |
|        17 | 14:10:46.3 | -16.019664 |      45561 |
|        18 | 14:10:46.6 | -15.843224 |      45493 |
|        19 | 14:10:47.0 | -15.666739 |      45403 |
|        20 | 14:10:47.3 | -15.490263 |      45103 |
|        21 | 14:10:47.6 | -15.313799 |      44882 |
|        22 | 14:10:47.9 | -15.137327 |      44508 |
|        23 | 14:10:48.3 | -14.960851 |      43949 |
|        24 | 14:10:48.6 | -14.784388 |      43580 |
|        25 | 14:10:48.9 | -14.607911 |      43376 |
|        26 | 14:10:49.3 | -14.431452 |      43177 |
|        27 | 14:10:49.6 | -14.254968 |      42776 |
|        28 | 14:10:49.9 | -14.078494 |      41944 |
|        29 | 14:10:50.3 | -13.902035 |      40380 |
|        30 | 14:10:50.6 | -13.725563 |      37200 |
|        31 | 14:10:50.9 | -13.549097 |      32607 |
|        32 | 14:10:51.3 | -13.372619 |      26837 |
|        33 | 14:10:51.6 | -13.196150 |      19778 |
|        34 | 14:10:51.9 | -13.019680 |      12157 |
|        35 | 14:10:52.3 | -12.843206 |       5154 |
+-----------+------------+------------+------------+
generator tune_dy ['76c24fef'] (scan num: 41)
                                                                                                                                                                       
I Wed-14:10:52 - PeakStats
================ ===============================
key              result                         
================ ===============================
x                d_stage_y                      
y                PD_USAXS                       
cen              -15.820231512268233            
com              -15.851207139335587            
fwhm             4.962447738943176              
min              [ -12.843206 5154.      ]      
max              [-1.6196144e+01  4.5831000e+04]
crossings        [-18.30145538 -13.33900764]    
tune_ok          True                           
center           -15.851207139335587            
initial_position -15.843211000000224            
final_position   -15.851207139335587            
================ ===============================

                                                                                                                                                                       
I Wed-14:10:53 - axis: d_stage_y
I Wed-14:10:53 - starting position: -15.843211000000224
I Wed-14:10:53 - peak detected: True
I Wed-14:10:53 -   max: (-16.19614400000023, 45831.0)
I Wed-14:10:53 -   center: -15.820231512268233
I Wed-14:10:53 -   centroid: -15.851207139335587
I Wed-14:10:53 -   fwhm: 4.962447738943176
Out[13]: 10:53 - final position: -15.851350000000252
('89322aeb-3f0c-46ae-a13d-2932f449d4ac',
 '76c24fef-dfec-44bd-ac04-128624753022')

```

prjemian, 2023-02-01:
Same code, two different devices: `d_stage.x` & `d_stage.y`  We'll compare those two devices first.

prjemian, 2023-02-02:
It's the device's `read_attrs` that were turned off by the uascan.  We changed to horizontal scattering geometry with this cycle, so we want dx but not dy.

prjemian, 2023-02-02:
https://github.com/APS-USAXS/usaxs-bluesky/blob/60a9eb7f2bea8306026734514b7ca650bf829b25/instrument/plans/uascan.py#L130-L133

commit 3dedbf9c (2023-02-02): FIX #583
closed 2023-02-02 by prjemian