APS-USAXS/usaxs-bluesky-ended-2023 #584: resetUSAXS broken by #583

jilavsky, 2023-02-03:
When we fixed #583 we forgot to restore for the motors parameters in resetUSAXS. When BS stops in uascan (due to anything) and does not finish, MR is not scannable and tune_mr fails. We need to add the fix from here to reset_USAXS 

https://github.com/APS-USAXS/usaxs-bluesky/blob/60a9eb7f2bea8306026734514b7ca650bf829b25/instrument/plans/uascan.py#L130-L133

jilavsky, 2023-02-27:
I have fixed this kind issue on usaxscontrol BS version (needs to be committed to GH), but we need to fix reset_USAXS running into suspender. At this time if we loose beam, RE(reset_USAXS) will fail due to suspender. That is not acceptable as we need to be able to reset the instrument even without beam and permissions.

jilavsky, 2023-02-28:
When I did RE(reset_USAXS()) BS went in suspender and was suspended. Which is confusing, since RE(mode_...()) works fine. And looking at reset_USAXS I do not see which line item causes the suspender to be engaged and BS suspending. We could remove the operation which is running into suspender.
We can test this next week Monday when we can reproduce same conditions. I will be bit late and have busy day, but we can find suitable time, may be issue line by line commands until we find what is suspending the BS.

jilavsky, 2023-02-28:
Sorry, hit the wrong button... Reopening again.

jilavsky, 2023-02-28:
Actually, found the log still:
```
n [11]: RE(reset_USAXS())
At least one suspender has tripped. The plan will begin when all suspenders are ready. Justification:
    1. Signal white_beam_ready_available is low
    2. Signal usaxs_CheckBeamStandard is low

Suspending... To get to the prompt, hit Ctrl-C twice to pause.
^CA 'deferred pause' has been requested. The RunEngine will pause at the next checkpoint. To pause immediately, hit Ctrl+C again in the next 10 seconds.
Deferred pause acknowledged. Continuing to checkpoint.
^Ctrying a second time
Pausing...
RunEngineInterrupted: 
Your RunEngine is entering a paused state. These are your options for changing
the state of the RunEngine:

RE.resume()    Resume the plan.
RE.abort()     Perform cleanup, then kill plan. Mark exit_stats='aborted'.
RE.stop()      Perform cleanup, then kill plan. Mark exit_status='success'.
RE.halt()      Emergency Stop: Do not perform cleanup --- just stop.

In [12]: RE.abort()
Aborting: running cleanup and marking exit_status as 'abort'...
Out[12]: ()
```

jilavsky, 2023-02-28:
Note, that both suspenders should be low, APS is off and shutters are closed, both are low.

prjemian, 2023-04-11:
See https://bcda-aps.github.io/apstools/dev/examples/ho_contingency.html#The-finalize_decorator()