APS-USAXS/usaxs-bluesky-ended-2023 #587: Improve on "specmacro.txt" for web site

jilavsky, 2023-02-10:
Our web site has "specmacro.txt" link which shows what usaxs.mac file content looks like. This is actually really useful for staff. 
BlueSky function run_my_command_file actually places list of collection commands there, so it is useful. 
When we run in BlueSky function (Linkam etc.) which collects data, this document does not get updated. It would be really cool, if in this case we put in the linked text file the function itself which is being run. I am now staring at user collecting linkam run and I have no idea what he is running. I would have to log into the control computer and even then, it will be challenge.
Alternative, at least put there name of function with parameters and python file where the function is, so I have a chance to figure this out.

prjemian, 2023-02-27:
I believe this is a duplicate of #462.  @jilavsky, do you agree?

jilavsky, 2023-02-27:
yes

closed 2023-02-27 by prjemian